dji-sdk/Onboard-SDK #230: M100 + Manifold, segmentation fault in sample application

adam3039, 2018-01-10:
I've got my manifold connected properly to the drone & DJI Go, with the drone connected to my PC for DJI Assistant. I'm successfully connecting to the drone, but getting a segmentation fault halfway through the application. Actions also aren't being reflected in the simulator, not sure if that is related.

ubuntu@tegra-ubuntu:~/Documents/OnboardSDK/build/bin$ ./djiosdk-flightcontrol-sample UserConfig.txt
Read App ID
User Configuration read successfully. 
STATUS/1 @ init, L55: Attempting to open device /dev/ttyTHS1 with baudrate 230400...
STATUS/1 @ init, L65: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L630: Device Serial No. = 041DE71664
STATUS/1 @ parseDroneVersionInfo, L632: Hardware = M100
STATUS/1 @ parseDroneVersionInfo, L633: Firmware = 3.1.10.0
STATUS/1 @ parseDroneVersionInfo, L636: Version CRC = 0xA6453AAC
ERRORLOG/1 @ functionalSetUp, L154: Subscriber not supported!
STATUS/1 @ functionalSetUp, L188: MFIO not supported!
ERRORLOG/1 @ functionalSetUp, L206: Hardware Sync not supported!
STATUS/1 @ activate, L1083: version 0x3010A00
STATUS/1 @ activate, L1096: Activation successful
| Available commands:                                            |
| [a] Monitored Takeoff + Landing                                |
| [b] Monitored Takeoff + Position Control + Landing             |
a
Aircraft hovering at 1064.45m!
Landing failed. Aircraft is still in the air.
Segmentation fault
ubuntu@tegra-ubuntu:~/Documents/OnboardSDK/build/bin$ ./djiosdk-flightcontrol-sample UserConfig.txt
Read App ID
User Configuration read successfully. 
STATUS/1 @ init, L55: Attempting to open device /dev/ttyTHS1 with baudrate 230400...
STATUS/1 @ init, L65: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L630: Device Serial No. = 041DE71664
STATUS/1 @ parseDroneVersionInfo, L632: Hardware = M100
STATUS/1 @ parseDroneVersionInfo, L633: Firmware = 3.1.10.0
STATUS/1 @ parseDroneVersionInfo, L636: Version CRC = 0xA6453AAC
ERRORLOG/1 @ functionalSetUp, L154: Subscriber not supported!
STATUS/1 @ functionalSetUp, L188: MFIO not supported!
ERRORLOG/1 @ functionalSetUp, L206: Hardware Sync not supported!
STATUS/1 @ activate, L1083: version 0x3010A00
STATUS/1 @ activate, L1096: Activation successful
| Available commands:                                            |
| [a] Monitored Takeoff + Landing                                |
| [b] Monitored Takeoff + Position Control + Landing             |
b
Aircraft hovering at 1064.44m!
Task timeout!
Task timeout!
Task timeout!
Landing failed. Aircraft is still in the air.
Segmentation fault

adam3039, 2018-01-11:
One of the issues was that it apparently didn't like that my PC (running assistant 2) was connected to the drone. I'm still getting the errors, but at least it's starting the motors now. The error while the drone was connected was "USB connected" in DJI Go. 

Is there any kind of "simulation" mode I should be looking for? I'd like to see flights reflected in the Simulator, but I'm not sure how I can connect DJI Assistant to the drone.

rohitsantdji, 2018-01-22:
Hi @adam3039 - for DJI Assistant simulation you need to go to the "Simulator" tab on the left, and then click the "Start Simulating" button.

adam3039, 2018-01-22:
Thanks @rohitsantdji , my issue was that the simulator popped up on top of the "Start Simulating" button. I didn't realize I needed to click that to actually begin simulating.

Cheers

closed 2018-01-23 by rohitsantdji