dji-sdk/Onboard-SDK #233: Cannot find Precise Mission Trajectory following library with latest version

gstream, 2018-01-23:
How to implement precise mission with latest Onboard SDK?

rohitsantdji, 2018-01-23:
Hi @gstream  - we don't have the precision missions suite available with the latest SDK, but to give you a sneak peek, we are working on an upgraded and more flexible version of this mission system. We're looking at supporting arbitrary waypoints from GPS or other sources of localization for this upgraded mission system. I don't have much more to share at the moment, but stay tuned! Do let us know if you have any feature requests.

QRI-nashw, 2018-01-23:
@rohitsantdji Hi Rohit, what is the difference between a precision mission vs. a normal one? When you mentioned the upgraded mission system, would that have an included moveByGPSOffset function? I think this would be a very useful and not-too-difficult function to implement.

gstream, 2018-01-23:
Hi @rohitsantdji , thanks for your update. I want to implement precision mission flying with M210-RTK. What is the best solution for it ? Thanks

rohitsantdji, 2018-01-24:
@QRI-nashw the upgraded mission system will have an interface that allows you to send GPS waypoints as well.

@gstream right now I don't think there is a ready solution, you would have to write your own code to generate the geometric shape and then use either position control, velocity control or attitude control APIs available in OSDK.

pingualoty, 2018-02-19:
This may be part of you answer above @rohitsantdji but having the ability to change the altitude of the drone at any time during a waypoint mission would be very useful.

Such as using a 3rd party height sensor and using that to control the height of the drone during a waypoint mission. Is that something that will be possible?

closed 2019-05-30 by hummelslei