dji-sdk/Onboard-SDK #234: [Wrong advanced-sensing library] Can not install djisdk-core on raspberry pi 3

DuongHieuMAI, 2018-01-25:
Hello, 

I can not install djisdk-core on my Rasp Pi 3. Problem is with advanced-sensing library at the step make djisdk-core:

- During the installation, process clones advanced-sensing-1.0.0.tar.gz file but it is required a file with version 2-0-0 (as in Ubuntu).

- If I replace the older version with the newer one. Command "make djisdk-core" is performed successfully but the next command is failed: "sudo make install djiosdk-core" with error: libadvanced-sensing.a format not recognized.

Please help me check if there were any problems.
Thank you very much.

Best reagards,
Drone Fan

SamuelWangDJI, 2018-01-25:
Hi
Can you provide the version of your Ubuntu?

- Rasp Pi 3 is armv7 architecture, which is not completely supported by Advanced-Sensing-2.0 due to FFMPEG driver issue. Please checkout the point No.4 in the "Limitation" section of this [page](https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-advanced-sensing-camera-stream.html). 

- Error message "xxx format not recognized" means the library you link is built on a different architecture, which leads to last question.

DuongHieuMAI, 2018-01-26:
Hi,
Thank you very much for your response.
- This is my Ubuntu version:

engineer@PC:~$ uname -a
Linux PC 4.13.0-31-generic #34~16.04.1-Ubuntu SMP Fri Jan 19 17:11:01 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
engineer@PC:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial
engineer@PC:~$ 

- This is the installation on my Rasp Pi 3:
pi@raspberrypi:~/Onboard-SDK $ mkdir build
pi@raspberrypi:~/Onboard-SDK $ cd build/
pi@raspberrypi:~/Onboard-SDK/build $ cmake .. -DADVANCED_SENSING=ON
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- External library - DJI Advanced Sensing for stereo cameras
Found FFMPEG FFMPEG_INCLUDE_DIR = /usr/include
Found FFMPEG FFMPEG_LIBRARIES   = /usr/lib/arm-linux-gnueabihf/libavcodec.so;/usr/lib/arm-linux-gnueabihf/libavformat.so;/usr/lib/arm-linux-gnueabihf/libavutil.so;/usr/lib/arm-linux-gnueabihf/libswscale.so;/usr/lib/arm-linux-gnueabihf/libvorbis.so;/usr/lib/arm-linux-gnueabihf/libvorbisenc.so;/usr/lib/arm-linux-gnueabihf/libtheora.so;/usr/lib/arm-linux-gnueabihf/libz.so
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so

stereo-vision-single-thread-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
-- OpenCV_INCLUDE_DIRS= /usr/local/include/opencv;/usr/local/include
-- OpenCV_LIBRARIES=opencv_xphoto;opencv_xobjdetect;opencv_ximgproc;opencv_xfeatures2d;opencv_tracking;opencv_text;opencv_surface_matching;opencv_structured_light;opencv_stereo;opencv_saliency;opencv_rgbd;opencv_reg;opencv_plot;opencv_optflow;opencv_line_descriptor;opencv_fuzzy;opencv_face;opencv_dpm;opencv_dnn;opencv_datasets;opencv_ccalib;opencv_bioinspired;opencv_bgsegm;opencv_aruco;opencv_videostab;opencv_videoio;opencv_video;opencv_superres;opencv_stitching;opencv_shape;opencv_photo;opencv_objdetect;opencv_ml;opencv_imgproc;opencv_imgcodecs;opencv_highgui;opencv_flann;opencv_features2d;opencv_core;opencv_calib3d

stereo-vision-multi-thread-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
-- OpenCV_INCLUDE_DIRS= /usr/local/include/opencv;/usr/local/include
-- OpenCV_LIBRARIES=opencv_xphoto;opencv_xobjdetect;opencv_ximgproc;opencv_xfeatures2d;opencv_tracking;opencv_text;opencv_surface_matching;opencv_structured_light;opencv_stereo;opencv_saliency;opencv_rgbd;opencv_reg;opencv_plot;opencv_optflow;opencv_line_descriptor;opencv_fuzzy;opencv_face;opencv_dpm;opencv_dnn;opencv_datasets;opencv_ccalib;opencv_bioinspired;opencv_bgsegm;opencv_aruco;opencv_videostab;opencv_videoio;opencv_video;opencv_superres;opencv_stitching;opencv_shape;opencv_photo;opencv_objdetect;opencv_ml;opencv_imgproc;opencv_imgcodecs;opencv_highgui;opencv_flann;opencv_features2d;opencv_core;opencv_calib3d

camera-stream-poll-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
-- OpenCV_INCLUDE_DIRS= /usr/local/include/opencv;/usr/local/include
-- OpenCV_LIBRARIES=opencv_xphoto;opencv_xobjdetect;opencv_ximgproc;opencv_xfeatures2d;opencv_tracking;opencv_text;opencv_surface_matching;opencv_structured_light;opencv_stereo;opencv_saliency;opencv_rgbd;opencv_reg;opencv_plot;opencv_optflow;opencv_line_descriptor;opencv_fuzzy;opencv_face;opencv_dpm;opencv_dnn;opencv_datasets;opencv_ccalib;opencv_bioinspired;opencv_bgsegm;opencv_aruco;opencv_videostab;opencv_videoio;opencv_video;opencv_superres;opencv_stitching;opencv_shape;opencv_photo;opencv_objdetect;opencv_ml;opencv_imgproc;opencv_imgcodecs;opencv_highgui;opencv_flann;opencv_features2d;opencv_core;opencv_calib3d

camera-stream-callback-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
-- OpenCV_INCLUDE_DIRS= /usr/local/include/opencv;/usr/local/include
-- OpenCV_LIBRARIES=opencv_xphoto;opencv_xobjdetect;opencv_ximgproc;opencv_xfeatures2d;opencv_tracking;opencv_text;opencv_surface_matching;opencv_structured_light;opencv_stereo;opencv_saliency;opencv_rgbd;opencv_reg;opencv_plot;opencv_optflow;opencv_line_descriptor;opencv_fuzzy;opencv_face;opencv_dpm;opencv_dnn;opencv_datasets;opencv_ccalib;opencv_bioinspired;opencv_bgsegm;opencv_aruco;opencv_videostab;opencv_videoio;opencv_video;opencv_superres;opencv_stitching;opencv_shape;opencv_photo;opencv_objdetect;opencv_ml;opencv_imgproc;opencv_imgcodecs;opencv_highgui;opencv_flann;opencv_features2d;opencv_core;opencv_calib3d
-- AdvancedSensing sample is enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Onboard-SDK/build
pi@raspberrypi:~/Onboard-SDK/build $ make djiosdk-core
Scanning dependencies of target advanced-sensing
[  2%] Creating directories for 'advanced-sensing'
[... cut ...]
pi@raspberrypi:~/Onboard-SDK/build/osdk-core/advanced-sensing-prefix/src/advanced-sensing $ ls
CMakeLists.txt  advanced-sensing-1.0.0.tar.gz

- If I replace this file with version 2.0.0 in Ubuntu, command "make djiosdk-core" is performed successfully, but next command "sudo make install djiosdk-core " is failed with error:  "libadvanced-sensing.a format not recognized" as I mentioned above.

- Could you help me check or let me know if there were any ways to install it on my Rasp?

Thank you for your help.
Best wishes to you,
DroneFan

rohitsantdji, 2018-02-05:
As @SamuelWangDJI mentioned, unfortunately currently we do not support AdvancedSensing 2.0.0+ on Raspberry Pi. Stay tuned and we will have compatibility soon!

closed 2018-02-05 by rohitsantdji

edancain, 2018-08-03:
@rohitsantdji, Rohit its Edan, do you guys have an update to this? 
Look forward to seeing you in Denver this October. Trying to get this to run on a Pi model 3B V1.2. Intention is to get this code working and then port to an Odroid XU4 running linux MATE as I have one here. Guidance module is here but the rest of the M100 is overseas including Manifold.

[ 16%] Performing build step for 'advanced-sensing'
[ 18%] No install step for 'advanced-sensing'
[ 18%] Completed 'advanced-sensing'
CMake Error: Problem with archive_read_open_file(): Failed to open '/home/pi/Onboard-SDK/build/osdk-core/advanced-sensing-prefix/src/advanced-sensing/advanced-sensing-2.0.1.tar.gz'
CMake Error: Problem extracting tar: /home/pi/Onboard-SDK/build/osdk-core/advanced-sensing-prefix/src/advanced-sensing/advanced-sensing-2.0.1.tar.gz
osdk-core/CMakeFiles/advanced-sensing.dir/build.make:125: recipe for target 'advanced-sensing' failed
make[3]: *** [advanced-sensing] Error 1
CMakeFiles/Makefile2:124: recipe for target 'osdk-core/CMakeFiles/advanced-sensing.dir/all' failed
make[2]: *** [osdk-core/CMakeFiles/advanced-sensing.dir/all] Error 2
CMakeFiles/Makefile2:99: recipe for target 'osdk-core/CMakeFiles/djiosdk-core.dir/rule' failed
make[1]: *** [osdk-core/CMakeFiles/djiosdk-core.dir/rule] Error 2
Makefile:162: recipe for target 'djiosdk-core' failed
make: *** [djiosdk-core] Error 2

Initially the command "cmake .. -DADVANCED_SENSING=ON" failed due to libusb not being installed on the Pi. Fix that and re-run the command. 
I got another error due to ximgproc and viz3D modules were missing from OpenCV. Install extra modules which took care of viz3D but it doesn't seem to have installed ximgproc so I get a failure when the stereo-vision-depth-perception-sample is being built.

stereo-vision-depth-perception-sample...
-- Found OpenCV 3.3.1 installed in the system, will use it for depth perception sample
--  - Includes: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab;opencv_viz;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_cvv;opencv_datasets;opencv_dpm;opencv_face;opencv_fuzzy;opencv_hdf;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_reg;opencv_rgbd;opencv_saliency;opencv_stereo;opencv_structured_light;opencv_surface_matching;opencv_text;opencv_tracking;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto
-- Did not find ximgproc in OpenCV
-- Found viz3d in OpenCV, will use it to visualize point cloud
-- Found OpenCV 3.3.1 and Viz3d module, stereo vision depth perception sample will be compiled.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Onboard-SDK/build

Regards,
Edan