dji-sdk/Onboard-SDK #236: Connection Issues - OSDK 3.5 on Ubuntu w. A3

cmacdonald33, 2018-01-26:
I'm running ubuntu on a virtual machine in order to feel out the OSDK before getting the SBC I'll be using for my project. As such, the A3 is connected to my PC via a USB-TTL connector as shown in the Hardware Setup. A3 has latest version of firmware running, and API control enabled in Assistant. Baud rate, port, app id and app key are all correct. Connections are correct. User has the correct dialout permissions.

As of yesterday, I was running the samples without issue. However, after reconnecting everything in the same exact configuration as yesterday, attempting to run the samples from terminal results in the following:

Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ parseDroneVersionInfo, L475: Drone version was not obtained. Please restart the program or call getDroneVersion

ERRORLOG/1 @ getDroneVersion, L1168: Drone version not obtained! Please do not proceed. 
Check if your serial connection is okay, and if your activation status is okay.

ERRORLOG/1 @ functionalSetUp, L139: Failed to initialize Version! Please exit.
Segmentation fault (core dumped)

I am unsure of how to deal with this issue and thus am turning here for help. Any advice would be much appreciated!
Cheers,
Calum

justwillim, 2018-01-28:
Mostly, ``` Drone version not obtained  ``` means that the serial port did not received any data.  PLZ check your hardware connection

cmacdonald33, 2018-01-29:
@justwillim How can I check my hardware connection beyond the fact that it is connected to the correct port?

cmacdonald33, 2018-01-29:
@justwillim Is a Lightbridge product also required for activation?

Jai-GAY, 2018-09-17:
i have the [same not good experience related to this](https://github.com/dji-sdk/Onboard-SDK/issues/374#issuecomment-421901219).

closed 2019-05-30 by hummelslei