dji-sdk/Onboard-SDK #240: Segmentation Fault Running from cron or rc.local

natelannan-osu, 2018-02-02:
I am currently trying to subscribe to some telemetry data and write to a file in a loop.  This works perfectly if I run the code manually from the terminal.  However, if I attempt to auto start using cron or rc.local the program runs correctly for an arbitrary amount of time (on average about 2 minutes) and then segfaults.  I would sincerely appreciate any suggestions you might have.  The code I have written is basically the same as the djiosdk-telemetry-example with an infinite loop getting values from the topics subscribed to and then writing these values to a file with a 50ms sleep in between loop iterations.

The hardware I have encountered this on is Beaglebone Black and Pocket Beagle.  I have tried running both Ubuntu 16 and 14.  

The flight controller I am using is an A3 running firmware 1.7.1.5 and the OSDK version I am using is 3.5

rohitsantdji, 2018-02-05:
That's quite interesting. We have run telemetry gathering code through cron and it has run without any issues for hours - however we have not tried printing to a file in any of these tests. Could you try your code without the file writing and let us know the results?

semihyildirim, 2018-02-05:
try some trivial stuff to give enough time to boot. For example; get system updates via "sudo apt-get update" and "sudo apt-get upgrade". Even without an internet connection, these will give you enough time to the system boot and your sample will start without taking an error.

natelannan-osu, 2018-02-06:
I was fooled into thinking the segfault was truncating my logs.  This is actually two issues, one of which I think has nothing to do with djiosdk.  The segfault is caused from a hard shutdown of the program (ie, not exited cleanly and the power is pulled).  The fact that my logs are not printing all events, is likely a buffer issue that I have not figured out yet.

natelannan-osu, 2018-02-07:
after further testing the segfault only occurs when the device has a hard shutdown (power pulled).

The file truncation was unrelated.  it was due to the linux write cache not getting cleared.  I put in  a sync() to fix that.  

Really this is an edge case that only happens when my device is forcefully shutdown.  Let me know if you want me to do anything else to investigate

Jai-GAY, 2018-09-13:
since like same [problem i encountered](https://github.com/dji-sdk/Onboard-SDK/issues/374).

closed 2019-05-30 by hummelslei