dji-sdk/Onboard-SDK #241: Load All Waypoints always fails!

kezhengtao, 2018-02-06:
Please refer to the attachment.
[LoadAllWaypointsFails.zip](https://github.com/dji-sdk/Onboard-SDK/files/1697389/LoadAllWaypointsFails.zip)

rohitsantdji, 2018-02-06:
Can you connect RC and mobile device, and let us know if there are any requests from DJI Go to sign in?

kezhengtao, 2018-02-06:
Yes ,I have connected RC. 
Please look at the content in the video,RC Status is P.

kezhengtao, 2018-02-06:
Is my  operation flow wrong?

rohitsantdji, 2018-02-06:
Your operation flow seems okay - does DJI Go ask you to sign in or provide any other prompts?

kezhengtao, 2018-02-06:
I just open RC in this test.Not open DJI go. I  think I can use QT instead of DJI Go in the test.
I will open DJI go next time to see prompts .

rohitsantdji, 2018-02-06:
I believe we found the cause of your error - In your workflow, you can avoid pressing the "Read Position as View Point" button, it is buggy. The "Add Waypoint" button should add the current drone location by default, even without pressing the "Read Position as View Point" button. Let us know if this fixes it.

kezhengtao, 2018-02-07:
Ye,You've just said a mouthful!Thanks

kezhengtao, 2018-02-08:
Could you tell me why?

rohitsantdji, 2018-03-06:
We would need to investigate further. Since your original problem (not being able to upload waypoints) has a workaround, I'm going to go ahead and close this issue.

closed 2018-03-06 by rohitsantdji