dji-sdk/Onboard-SDK #242: can't communicate with M210 using the USB-TTL cable.

Alikhajehei94, 2018-02-06:
I have followed the DJI guide in running a sample app from their library and this when I get when I run the sample. "Drone version not obtained"
[image]

Any idea why this is happening?
how can I make sure to see if the linux is able to communicate with the M210?
Thanks,
Ali

rohitsantdji, 2018-02-08:
Did you try all the suggestions the error message is giving you?

Alikhajehei94, 2018-02-08:
Yes I tried all the massages displayed. Still thesame error.

rohitsantdji, 2018-02-08:
Could you share screenshot of your Assistant 2 SDK Page and a picture of your serial connections?

Alikhajehei94, 2018-04-06:
I figured what it was wrong. I had a faulty cable.

closed 2018-04-06 by Alikhajehei94