dji-sdk/Onboard-SDK #243: Auto take off

Vladilen, 2018-02-07:
Hello!
We're trying to take off with N3, but it mode switches to MODE_ASSISTED_TAKEOFF. How can we get MODE_AUTO_TAKEOFF?
The steps are:
1. Activation (cmd_set 0x0, cmd_id 0x1) is ok.
2. Verify (cmd_set 0x0B, cmd_id 0x0) is ok. 
3. Parsing telemetry, configured from DJI Assistance is ok. We don't subscribe to telemetry from board soft.
4. Obtain control authority (cmd_set 0x1, cmd_id 0x0) is ok (first answer is 0x03, second is 0x02).
5. Trying to do takeoff by switching fly mode (cmd_set 0x1, cmd_id 0x1) using OSDK 3.3+ command type 0x1. Copter arms only with RC turned on. Throttle controlled only from RC, and copter takes off when throttle on RC is greater 50%. Mode 41 on arming and 10 on takeoff, then it returns to 6. 10 is MODE_ASSISTED_TAKEOFF, but we want to MODE_AUTO_TAKEOFF (0x11). How can we can get auto takeoff mode and how can we control takeoff altitude from OPEN protocol?

rohitsantdji, 2018-02-08:
For (5), this is not expected behavior. On the DJI Assistant 2 SDK page, there is a setting for Enable SDK failsafe, can you turn that ON? Additionally, can you select the RC lost action to "Hover"?

Vladilen, 2018-02-08:
Yes, we enable failsafe on SDK page. The action was "landing". When we changed it to "Hover", copter began to arm without RC when we send takeoff command, but the mode still MODE_ASSISTED_TAKEOFF.
Yesterday we tried missions. After obtaining control authority we send WP Init (cmd_set 0x3, cmd_id 0x10), then uploading two points (cmd_set 0x3, cmd_id 0x11). After this, when on ground, send start mission (cmd_set 0x3, cmd_id 0x12). Mode changes to MODE_AUTO_TAKEOFF, copter began to takeoff without controlling with RC. But after it has altitude 1 meter - copter began landing and didn't fly in waypoint.
If we start mission in the air, when altitude is 25 meters, copter first fly down 10-15 meters and then fly to waypoint. If we start mission on another altitude - the behavior is same - it looses 10-15 meters altitude and the fly to WP. In WP init command we tried set focus altitude to current copter altitude and tried to set it fixed to 20 meters, but no changes in behavior. Why can it be?
We control N3 from own GCS, and have STM32 on board. WP stores in our controller and we just want from N3 to fly in WP that we transmit to it. Waypoints can change randomly, but if we will use mission then after changing WP we must reinitialize whole mission -> init - upload - upload - start. Hotpoint works fine, but it doesn't support 0 radius to stop in point coordinates, it can stop in 5 meters from it. As we found about "follow me" - it works only in 200 meters around copter. And we can't start follow me - copter didn't hovers in place.
Landing command (cmd_set 0x1, cmd_id 0x1) using OSDK 3.3+ command type 0x2 doesn't work in all tried cases.
Can you please explain the easy way to control N3 how we want: auto take off to some altitude, flying to changing coordinates, landing?

closed 2018-02-27 by Vladilen