dji-sdk/Onboard-SDK #245: Non-invasive Hotpoint

QRI-nashw, 2018-02-08:
It appears that in order to execute a hotpoint, you first need to position the drone at the center of the the hotpoint location. From there, the drone moves itself to the radius and begins rotating. Is there a way to execute a hotpoint with a drone already positioned at the radius of the hotpoint? So instead of the flight plan being:: 
1. Fly to center of hotpoint
2. Fly to perimeter of hotpoint
3. Begin rotation
4 Stop rotation, fly way

It would be ::
1. Approach hotpoint location, but stop at a point along the perimeter (1 radius length from center)
2. Begin rotation
3. Stop rotation, fly away

Thanks!

rohitsantdji, 2018-02-14:
Interesting suggestion. One use case I see is orbiting infrastructure without actually hitting it. Is your use case something like this? 

There are no plans to implement this right now, but I'll cross reference this to #233 as with some of the other suggestions, and we'll pick it back up in a few months.

closed 2018-02-14 by rohitsantdji

QRI-nashw, 2018-02-14:
Yes @rohitsantdji , exactly. To be honest, I will probably work on a program to accomplish this, as I'd like to get this function working sooner than a few months. I'll keep ya'll updated.