dji-sdk/Onboard-SDK #246: DJI A3 problems with SDK > 3.3

OShapovalov, 2018-02-09:
When we use e6c293341fb5461e8b364a584ca2bae8855cb5d5 (3.3) revision for onboard-sdk and ffc2f08a7f5da0f7d590fcba281d80706dd975cd for onboard-sdk-ros, it works.

If we update both sdk to 3.4 or 3.5 we have NaN instead A3 voltage.

When we update to 3.6 - we get error "Vehicle initialization failed" during dji-sdk ros node start.

We hope that this problem will be fixed.

rohitsantdji, 2018-02-20:
Do you still see the 3.6 "Vehicle initialization failed" error with the latest 3.6.1 release?

For the voltage, we'll look into it and if there is a fix to be made, we will roll it into our next release.

OShapovalov, 2018-02-26:
Now we have NaN instead A3 voltage (with latest 3.6.1 release).

OShapovalov, 2018-04-02:
Can we expect fix for this problem?

closed 2019-05-30 by hummelslei