dji-sdk/Onboard-SDK #247: OSDK landing issue

ASoleev, 2018-02-09:
Hello!

Issue1: 

We are using Missions from Onboard SDK with finish action AUTO_LANDING on mission exit. 
Is it possible to stop landing process by any command? 

Issue2:

Is is possible to perform auto landing process without Waypoint mission? For example, in case of Hotpoint mission.

rohitsantdji, 2018-02-15:
Issue 1:

You can call `vehicle->control->action()` with a parameter `FlightCommand::exitTakeOff`. [Here](https://developer.dji.com/onboard-api-reference/classDJI_1_1OSDK_1_1Control.html#a5e8df1234fe307c01d209b67f34c9efe) is the API reference for that.

Issue 2: 

You can set up your code to call `vehicle->control->action()` with the parameter `FlightCommand::Landing` whenever you need to land. You could wait for the hotpoint mission to complete and then call this function to execute landing.

rohitsantdji, 2018-02-20:
Closing since no activity from OP. Please re-open this issue if you have additional follow-ups.

closed 2018-02-20 by rohitsantdji

sharvashish, 2018-03-21:
Hello @rohitsantdji ,

Are the exitTakeOff, exitLanding, exitGoHome APIs supported by A3/N3/M100/M600?
Is there any vehicle which does not support this API?