dji-sdk/Onboard-SDK #250: Takeoff/Landing Sample Function Issues

QRI-nashw, 2018-02-12:
For the past week or so I've  been running into some strange problems when trying to land and takeoff with the drone. Initially, I was having problems using my own functions, so I decided to test the sample functions provided with the SDK to see if the same issue occurred, and sure enough, a similar problem keeps happening. You can see what happens in this video: https://www.youtube.com/watch?v=Y97NazcPLaI

Basically, after starting the simulator and opening djiosdk-flighcontrol-sample, pressing "a" does nothing. So I press ctrl+C to exit the sample and retry. This time, the drone takes off, but when it attempts to land, I get an error, as you can see in the video. 
The errors are as follows:

STATUS/1 @ getErrorCodeMessage, L644:
STATUS/1 @ getCMDIDTaskMSG, L823: CONTROLLER_FAIL

Any idea what could be causing this?

QRI-nashw, 2018-02-12:
Forgot to add, using an M100.

rohitsantdji, 2018-02-14:
Two suggestions:

1. Could you change your simulated co-ordinates to a GPS location outside of mainland China, unless you are physically located in China?

2. We could reproduce your issue with the sample partly, one possible cause is insufficient sleep time in one of the monitoring functions. in flight_control_sample.cpp, could you change L261 from `sleep(3)` to `sleep(4)`?
[image]

We'll be interested in hearing about your results.

QRI-nashw, 2018-02-14:
Changing my GPS coordinates to a spot in the US didn't seem to have an effect, but increasing the delay by a second seemed to fix the issue! Thanks @rohitsantdji ! It also gives an accurate readout for the "aircraft hovering at x" line too. Before it would always display "hovering at .1m" but now it gives an accurate number.