dji-sdk/Onboard-SDK #252: Takeoff/Landing Sample  Not Working

Merwanski, 2018-02-13:
I am trying to run the sample ./djiosdk-flightcontrol-sample UserConfig.txt  
Basically, after starting the simulator and opening djiosdk-flighcontrol-sample, I can get some information from the drone such as the version but after pressing "a" I get the following errors:

STATUS/1 @ getErrorCodeMessage, L644: a
STATUS/1 @ getCMDIDTaskMSG, L823: CONTROLLER_FAIL

STATUS/1 @ getErrorCodeMessage, L644: a
STATUS/1 @ getCMDIDTaskMSG, L823: CONTROLLER_FAIL

Any idea what could be causing this? an how can I fix it

QRI-nashw, 2018-02-13:
Hey Merwanksi, check out issue #250 , is it describing the same problem?

rohitsantdji, 2018-02-14:
Please tell us what drone you are using and what SDK version you are using.

Merwanski, 2018-02-14:
@QRI-nashw I checked the issue #250  it is the same problem that you had, did you manage to find a solution for it?

@rohitsantdji we are doing the test on DJI M100 and we are using the SDK 3.4

QRI-nashw, 2018-02-14:
@Merwanski I figured it had to be, we used essentially the same wording haha, eerily similar. But no, I was not able to find a solution. I am using an M100 and SDK 3.5. I did realize, however, that this error only occurs when trying to execute landing directly after takeoff or vice versa. If you split the two commands up with a moveByPositionOffset or Hotpoint, no error is thrown. I have a feeling it has something to do with the "cleanup" block of code I've seen around the SDK, or perhaps data subscription/broadcast, but I can't know for sure. Hopefully @rohitsantdji can help us out.

QRI-nashw, 2018-02-14:
@Merwanski Scratch that, got it fixed! #250

Merwanski, 2018-02-15:
@QRI-nashw , thank you for the update, for my case I was using the M100 with the SDK 3.4.
I will try the SDK 3.5 with this solution that you proposed and see if I still get the error or not.

Merwanski, 2018-02-16:
After a trial yesterday with the SDK 3.5 and with/without sleep time solution as proposed in the #250 
I am still getting the same errors when I use the "djiosdk-flighcontrol-sample"

STATUS/1 @ getErrorCodeMessage, L644: a
STATUS/1 @ getCMDIDTaskMSG, L823: CONTROLLER_FAIL

STATUS/1 @ getErrorCodeMessage, L644: a
STATUS/1 @ getCMDIDTaskMSG, L823: CONTROLLER_FAIL