dji-sdk/Onboard-SDK #257: Does DJI store waypoints in SD card?

sharvashish, 2018-02-19:
Hi All,

I am able to set and get waypoints using your SDK.
But if I set a waypoint in DJI, and then restart your SDK I am unable to fetch waypoints.

Moreover, If I restart DJI hardware will it still preserve previously set waypoints?

rohitsantdji, 2018-02-20:
1) For the issue where restarting the SDK does not let you fetch waypoints, you might have to first initialize a MissionManager object with a MissionType::WAYPOINT argument after you restart. 

2) For the question about restarting the hardware, waypoints are stored in volatile memory in the flight controller. So they will not be preserved after a restart of the hardware.

sharvashish, 2018-02-20:
Hi @rohitsantdji ,
Yes, you are correct. I was indeed able to fetch waypoints once I initialized MissionManager object.
Thank you for clarifying that waypoints are indeed stored in volatile memory!

closed 2018-02-23 by rohitsantdji