dji-sdk/Onboard-SDK #258: How do I get height above MSL from TOPIC_ALTITUDE_FUSIONED?

rpici, 2018-02-20:
Based on this reference:

https://github.com/dji-sdk/Onboard-SDK/wiki/Altitude-Reference-for-different-topics

the altitude returned from TOPIC_ALTITUDE_FUSIONED is an ICAO height, which quite often is negative.

Is it that case that, in order to convert the ICAO value from TOPIC_ALTITUDE_FUSIONED to height above MSL, we are required to know the current barometric pressure at sea level in our specific location?  Is this correct?

rohitsantdji, 2018-02-20:
Yes, this is correct. Additionally, barometric calibrations differ from unit to unit up to a few meters.

rpici, 2018-03-21:
My problem is that I'm taking the altitude value from TOPIC_ALTITUDE_FUSIONED and (incorrectly) directly storing it as GPS altitude above MSL as metadata in my application.   When a user views the metadata, seeing a negative altitude above MSL makes no sense.

So to clarify, to get the current barometric pressure at sea level in our location, we need to rely on an outside source, e.g. a weather report.  Is this correct?

Is there a way to more conveniently convert TOPIC_ALTITUDE_FUSIONED to altitude above MSL, preferably one that doesn't rely on information from an outside source, i.e. a current local weather report?

Actually, altitude above MSL doesn't really have much meaning to an end user either.  An end user would be more interested in height AGL, i.e. the "Homepoint Altitude" mentioned [here](https://github.com/dji-sdk/Onboard-SDK/wiki/Altitude-Reference-for-different-topics).

I guess one way to achieve that now, without waiting for an SDK update, would be to record the TOPIC_ALTITUDE_FUSIONED at the homepoint before takeoff, and then use that as the baseline for all subsequent altitudes retrieved from TOPIC_ALTITUDE_FUSIONED, i.e. subtract subsequent altitudes from the homepoint altitude to get height above takeoff point, and then store that difference in my metadata.

What do you think of this approach?

closed 2019-05-30 by hummelslei