dji-sdk/Onboard-SDK #259: DJI M100 -  Takeoff/Landing Sample Not Working with the OSDK 3.6

Merwanski, 2018-02-20:
I came back to you because even after trying the latest OSDK 3.6 I am still getting the same error as in the previous issue  #252 each time I try to run the example djiosdk-flightcontrol-sample.
The full output I am getting is the following:

_**./djiosdk-flightcontrol-sample 
Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...
STATUS/1 @ init, L65: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L687: Device Serial No. = 041DE51233
STATUS/1 @ parseDroneVersionInfo, L689: Hardware = M100
STATUS/1 @ parseDroneVersionInfo, L690: 
STATUS/1 @ parseDroneVersionInfo, L693: Version CRC = 0xA6453AAC
STATUS/1 @ initSubscriber, L738: Telemetry subscription mechanism is not supported on this platform!
STATUS/1 @ initMFIO, L941: MFIO is not supported on this platform!
STATUS/1 @ initHardSync, L1003: Hardware Sync is not supported on this platform!
STATUS/1 @ activate, L1239: version 0x3010A00
STATUS/1 @ activate, L1252: Activation successful
| Available commands:                                            |
| [a] Monitored Takeoff + Landing                                |
| [b] Monitored Takeoff + Position Control + Landing             |

a

STATUS/1 @ getErrorCodeMessage, L644: 
STATUS/1 @ getCMDIDTaskMSG, L824: CONTROLLER_FAIL

STATUS/1 @ getErrorCodeMessage, L644: �9Gz
STATUS/1 @ getCMDIDTaskMSG, L824: CONTROLLER_FAIL**_

Do you have any idea what is causing this error and how I can fix it ?

rohitsantdji, 2018-02-20:
Some questions:

1. What is the simulation location you are inputing?
2. Is DJI Go connected and showing no errors?

Merwanski, 2018-02-20:
Answers 
1. In the simulation, I am using the following GPS coordinate (Lat, long) = (50.86, 4.68)
2. Yes the DJI is connected and I have no errors

rohitsantdji, 2018-02-23:
Your simulated location might be in a no-fly zone - could you try simulating at (37.4118, -121.9986) and report the results?

Merwanski, 2018-02-23:
I keep getting the same error 

STATUS/1 @ getErrorCodeMessage, L644: 35C�
STATUS/1 @ getCMDIDTaskMSG, L824: CONTROLLER_FAIL

Meowster808, 2018-06-14:
is your controller mode set to F?

azimgivron, 2019-05-20:
Hi, 
Could you tell me how you fixed it ? I have the same issue(same message error), I am also running on an M100 and the remote controller is set to F but it is still not working. I have no ideas where to look for.

Merwanski, 2019-05-24:
Hi 

@Meowster808 Thank you for your tip it was fixed at that time by setting the controller to F

@azimgivron Check also the configuration of the drone in "DJI Assistant 2" if I remember correctly, in addition, to set the controller to F you should also update some parameters in the software (but I don't remember exactly what was it :( sorry).
Let me know if you find it

Merwan

azimgivron, 2019-05-25:
Hi,

Thank you for your reply, I actually forgot to ask explicitly for authority in my software, I though that the takeoff command would do it by itself but this is not the case(the controller was already set to F).

Regards,
Givron Azim

closed 2019-05-30 by hummelslei