dji-sdk/Onboard-SDK #260: Getting video onboard

AutisticNinja, 2018-02-20:
I'm working on an application with an M600 and need to get the video feed from the X3/Z3 camera to an onboard device for processing.  The manifold seems to be ideal for this application, but there are some major problems:

The manifold-cam project here:
https://github.com/dji-sdk/Manifold-Cam
Seems to be abandoned.  No meaningful commits in the past 2 years.  Support questions go unanswered, including this one:
https://github.com/dji-sdk/Manifold-Cam/issues/15
which is exactly like a problem I have encountered.

The manifold seems to have been designed for only the M100.  In uses 8-pin connectors for the cam in and cam out, while the gimbals on the M600 use 7 pin cables.  Is there a way to connect the manifold to the camera with the M600 wiring?  

Is there any other way to get video from a dji gimbal to an onboard device?

rohitsantdji, 2018-02-23:
@AutisticNinja you are correct, the Manifold's video feeds are only doable with the M100 because of the 8pin/7pin connector. The M600 uses LightBridge 2 v/s LightBridge 1 for the M100, and the cables are different for both of these. 

The Onboard SDK currently supports video feeds through USB on the Matrice 210, no special cables required. However for the M600 there is currently no way to do so.

rcabg, 2018-03-22:
Hi @rohitsantdji , 

correct me if I'm wrong, but I'm trying to understand how getting video onboard works in order to apply computer vision:

- M210 + Linux OnBoard Computer (TX2, Manifold upgraded to 16.04....) -> "Advanced sensing (osdk 3.5+)" could be used. Resolution from main camera can be 1080p or 720p on video mode at 30 fps. Just Zenmuse x4s and Zenmuse x5s cameras work with this feature.

- M100 + Manifold (no possible other linux pc) -> Onboard-SDK-ROS 3.2 could be used. Resolution from main camera is 640x480 always. Just X3 camera works. 

 - Rest of DJI drones (M600, ...): so far no possible to get onboard video stream.

Am I right? Is there any other possibility for doing so?

Thanks in advance.

hummelslei, 2019-05-30:
Close this issue now. If you still have more issues related to DJI SDK, please help to fill in this form: 
https://formcrafts.com/a/dji-developer-feedback-en to report them to DJI Developer Support. Thanks for your understanding and cooperation.

closed 2019-05-30 by hummelslei

pavankumarbn, 2019-06-05:
Hi, I have tried to get Zenmuse X3 video feed (Mounted Matrice 100) on Windows computer using RTMP protocol and OBS (Also works on Linux). It worked perfectly fine for me.  
If you find any other alternative solution please update.