dji-sdk/Onboard-SDK #267: Invalid GPS data received in running telemetry sample

yekaifeng, 2018-02-26:
Rspi3 connects to the DJI N3 controller api ports, open field, blinking LED shows GPS status is good. While I was executing the sample/linux/missions/mission_sample.cpp in pi3, the GPS data received is very confusing -- at the sea in Africa! I believe these data is invalid. Any idea on that?

Below are the related info:

- Controller: N3
- Compass: GNSS-compass
- Firmware: v1.7.1.5
- OSDK: v3.3
- Assistant2: v1.1.8
- Serial Baud Rate: 115200
- OSDK platform: Raspi 3, ubuntu linux
[image]

```
STATUS/1 @ init, L37: Attempting to open device /dev/ttyUSB0 with baudrate 115200...

STATUS/1 @ init, L47: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L568: Device Serial No. = 0DBDE67002000X

STATUS/1 @ parseDroneVersionInfo, L570: Hardware = N3

STATUS/1 @ parseDroneVersionInfo, L571: Firmware = 3.2.36.8

STATUS/1 @ startPackage, L314: Start package 0 result: 0.
STATUS/1 @ startPackage, L316: Package 0 info: freq=50, nTopics=1.
STATUS/1 @ removePackage, L423: Remove package 0 successful.
ERROR/1 @ initGimbal, L735: Gimbal not mounted!

ERROR/1 @ functionalSetUp, L157: Failed to initialize Gimbal!

ERROR/1 @ functionalSetUp, L187: Virtual RC not supported!

STATUS/1 @ activate, L951: version 0x3022408

STATUS/1 @ activate, L964: Activation successful
| Available commands:                                            |
| [a] Get telemetry data and print                               |
a

STATUS/1 @ startPackage, L314: Start package 0 result: 0.
STATUS/1 @ startPackage, L316: Package 0 info: freq=1, nTopics=1.
STATUS/1 @ startPackage, L314: Start package 1 result: 0.
STATUS/1 @ startPackage, L316: Package 1 info: freq=10, nTopics=2.
STATUS/1 @ startPackage, L314: Start package 2 result: 0.
STATUS/1 @ startPackage, L316: Package 2 info: freq=50, nTopics=2.
STATUS/1 @ startPackage, L314: Start package 3 result: 0.
STATUS/1 @ startPackage, L316: Package 3 info: freq=200, nTopics=1.Counter = 0:
-------
Flight Status                         = 204
Position              (LLA)           = 0.403353, 1.97668, 37.5086
RC Commands           (r/p/y/thr)     = 14, 58, 28, -10000
Velocity              (vx,vy,vz)      = 9.49683e-05, -0.00137781, 0.024491
Attitude Quaternion   (w,x,y,z)       = 0.00195523, -0.00843607, -0.0167853, -0.999822
-------
```

yekaifeng, 2018-02-26:
The simulator is NOT on in DJI assistant2 .

rohitsantdji, 2018-02-26:
GPS Data is in radians, not degrees - multiply by (180/pi) to get your location

closed 2018-02-26 by rohitsantdji

yekaifeng, 2018-02-27:
True, it's radians. Thanks a lot!