dji-sdk/Onboard-SDK #271: Potential Code hangup when calling obtainCtrlAuthority(int timeout)

sharvashish, 2018-03-08:
https://github.com/dji-sdk/Onboard-SDK/blob/3.6/osdk-core/api/src/dji_vehicle.cpp#L1836-L1840

ACK::ErrorCode
Vehicle::obtainCtrlAuthority(int timeout)
{
  ACK::ErrorCode ack;
  uint8_t        data = 1;

  protocolLayer->send(2, this->encrypt,
                      OpenProtocolCMD::CMDSet::Control::setControl, &data, 1,
                      500, 2, false, 0);

  ack = *(ACK::ErrorCode*)waitForACK(
    OpenProtocolCMD::CMDSet::Control::setControl, timeout);

  if (ack.data == OpenProtocolCMD::ErrorCode::ControlACK::SetControl::
  OBTAIN_CONTROL_IN_PROGRESS)
  {
    ack = this->obtainCtrlAuthority(timeout);
  }

  return ack;
}

In this function, 
 if (ack.data == OpenProtocolCMD::ErrorCode::ControlACK::SetControl::
  OBTAIN_CONTROL_IN_PROGRESS)
  {
    ack = this->obtainCtrlAuthority(timeout);
  }
might result into a potential code hangup. It might be possible that the condition remains always 'True'.
I am being able to replicate this scenario, when I am executing a mission via DJI GO app, and then I call this API. It simply does not respond till waypoint mission gets over.

Is it possible to limit the number of times this recursive loop gets called!

rohitsantdji, 2018-03-16:
That's good feedback, we will fix this in the next release. However there is currently no short-term plan for a release so you might have to wait for about a month, perhaps a little more.

ol-bones, 2018-11-09:
Is this fixed in 3.7?

closed 2019-05-30 by hummelslei