dji-sdk/Onboard-SDK #272: Bus error occurs when subscribing to both TOPIC_VELOCITY and TOPIC_ALTITUDE_FUSIONED @200Hz

natelannan-osu, 2018-03-08:
My setup:  
Beagle bone black running ubuntu and OSDK 3.6
M600 running V1.00.01.65

Subscription list that caused the error:
@10  Hz: TOPIC_STATUS_FLIGHT
@50 Hz: TOPIC_GPS_FUSED, TOPIC_GPS_DATE,
			TOPIC_GPS_TIME, TOPIC_GPS_DETAILS, TOPIC_BATTERY_INFO,
			TOPIC_CONTROL_DEVICE, TOPIC_RC
@100 Hz:  TOPIC_COMPASS
@200 Hz:  TOPIC_QUATERNION, TOPIC_ACCELERATION_BODY,
			TOPIC_ANGULAR_RATE_FUSIONED, TOPIC_VELOCITY,
			TOPIC_ALTITUDE_FUSIONED

The error occured when I attempted to get an altitude value:
altitude = vehicle->subscribe->getValue<TOPIC_ALTITUDE_FUSIONED>();

I was successful in getting data from flight status, gps fused, gps date, gps time, and gps details before I hit this error.  If I moved TOPIC_ALTITUDE_FUSIONED to a different frequency I still hit the error, as well as if I switched to TOPIC_ALTITUDE_BAROMETER.  If I commented out my subscription to either TOPIC_VELOCITY or TOPIC_ALTITUDE_FUSIONED, my code would run fine.  If I tried commenting out anything else, I still hit the problem.

When I switched TOPIC_VELOCITY to 100 Hz the code ran fine.  Is there any restrictions to what and how many things you can subscribe to at once?

hummelslei, 2019-05-30:
Close this issue now. If you still have more issues related to DJI SDK, please help to fill in this form: 
https://formcrafts.com/a/dji-developer-feedback-en to report them to DJI Developer Support. Thanks for your understanding and cooperation.

closed 2019-05-30 by hummelslei

rskelly, 2019-11-19:
Had the same issue. Moved the velocity subscription into 50Hz to "fix."