dji-sdk/Onboard-SDK #273: APP would crash if we set multiple missions

magenious, 2018-03-09:
We used v3.3 OSDK in our app, and in our program we only set 2 waypoints in one waypoint mission, but in one flight, we may have a lot of missions(that is, we set mission one by one in one flight), then we find that the app would crash sometimes, the crash was hard to catch and no dump file generated, the app just crash  with exit code 3221225477 (0xC0000005) or 2147483651. 
After analysis, we find that issue should be caused by array access violation, for example, the constant variable MAX_MISSION_SIZE was set to 5 in dji_mission_manager
`private:
  //! @note no dynamic container, so fix the size of the mission container
  static const int MAX_MISSION_SIZE = 5;

  WaypointMission* wpMissionArray[MAX_MISSION_SIZE];
  HotpointMission* hpMissionArray[MAX_MISSION_SIZE];`

then in waypoint mission initialization in dji_mission_manager.cpp:
`void
MissionManager::initWayptMission(VehicleCallBack callback, UserData wayptData)
{
  wpMissionArray[wayptCounter] = new WaypointMission(this->vehicle);
  wpMission                    = wpMissionArray[wayptCounter];
  wayptCounter++;

  wpMission->init((WayPointInitSettings*)wayptData, callback, wayptData);
}`

we could find that if the count of waypoint mission was greater than 4， the access for wpMissionArray would be violated, then the sdk would possibly crash.
Indeed, we modified the value of MAX_MISSION_SIZE from 5 to 1000, then in our usage(set waypoints mission one by one in one flight), the crash frequency was decreased a lot, but unfortunately, it still crash   at times, and the crash symptom was the same as before.
Since the debugging for array access violation was hard(it's hard to reproduce the crash and no dump file generated for heap corruption), we still investigate about the issue, would you give some tips about **which arrays would cause access violation in our usage**? Greatly appreciate for your help.

rohitsantdji, 2018-03-16:
Can you send us a working piece of code that causes this crash even when the MAX_MISSION_SIZE is set to 1000? If you do not want to share your full code then a minimal example that can reproduce this crash will also do. Please email this to dev@dji.com and reference this Github issue number.

magenious, 2018-03-21:
Hi, @rohitsantdji 
   I have sent the email, thanks for your great help.

tompe17, 2018-05-04:
I set MAX_MISSION_SIZE to 5000 to avoid this crashing.  Setting this to a higher value as default would be really useful. Or being able to specify it as a parameter.

closed 2019-05-30 by hummelslei