dji-sdk/Onboard-SDK #274: SDK Failsafe missing in DJI Assistant 2 with M100

rocklinsuv, 2018-03-09:
Hi there,

I've been playing with my M100 recently, and managed to use DJI Assistant 2 (both v1.1.6 and v1.2.0) with M100 firmware version 1.3.1.00 to tune the telemetry frequency. However, it seems that the "Enable SDK Failsafe" box is missing here:
[image]
No matter how I selected the action, it reset to nothing all the time. Is there anything I can do to fix this? Any help would be very much appreciated!

Cheers,

rohitsantdji, 2018-03-16:
Unfortunately, SDK Failsafe action is not supported on M100.

closed 2018-03-16 by rohitsantdji

rocklinsuv, 2018-03-17:
I believe that flicking the switch from "F" to either "A" or "P" on the RC will terminate the SDK and re-enable manual control. Can this be considered as some sort of "Failsafe"? Correct me if I'm wrong, thanks!

rohitsantdji, 2018-03-17:
The term "SDK Failsafe" is a little misleading - it actually means that when the RC is lost, and the drone enters failsafe mode, the SDK can take over. So what you mentioned is true, but it is a little different from the meaning of SDK Failsafe on our other aircraft.

rocklinsuv, 2018-03-17:
Sure. At least there is a way to take back control when things go south for M100. That's all what I need. Thanks again for your help!

QRI-nashw, 2018-03-19:
@rohitsantdji So using the M100, if you want to execute a pre-programmed mission with the SDK, you must have a connection to the RC? What would losing that connection mid-mission look like?