dji-sdk/Onboard-SDK #276: Motors not alway starting

dimitrios1988, 2018-03-12:
After compiling and running one of your samples, I noticed an issue with the motors when not attached to a simulating environment. The first time, the motors start normally, but when I stop the program and execute it again, the motors don't start. I also checked the returned variable from the Control::takeoff(int wait_timeout) function, but it returns SUCCESS, even when the motors haven't started.

When I run the sample program with the simulator, it works all the times. The problem occurs only when it is not in the simulator.

I am using a Matrice100 with the latest firmware (1.3.1.00) and the latest OSDK version.

rohitsantdji, 2018-03-16:
Do you turn off the motors manually with the remote controller, or with an API call?

dimitrios1988, 2018-03-16:
Manually with the remote controller. Does it matter?

rohitsantdji, 2018-03-17:
We've seen this behavior when the motor on/off is asymmetric i.e. SDK turns motors ON, RC turns them OFF or the other way around. Can you attempt this with purely SDK-based ON/OFF?

dimitrios1988, 2018-03-17:
Ok I will try it on Monday and I will let you know.

QRI-nashw, 2018-03-19:
Thank you for posting this @dimitrios1988, I've been experiencing this issue for a month or two now, even in the simulation environment. Typically, if you exit a sample while the craft is in simulated flight, and try to execute a takeoff command after resetting the simulator, the drone will display the behavior @dimitrios1988 described. After exiting the sample again and re-executing the takeoff command, the drone behaves as intended. After some testing, it seems that the drone behaves this way in the field as well. I'm usually turning off the motors with an API call, only with the controller if the drone becomes unresponsive.

QRI-nashw, 2018-03-19:
Another interesting thing to note is that if the drone is allowed to completely finish a mission (i.e. takeoff, move to waypoints, and return home), it seems to have no issue taking off immediately with a single API call. If however, you need to interrupt the mission, it typically takes a takeoff API call + exit sample + another API takeoff call to get the drone to respond. But as @dimitrios1988 noted, the flight controller returns a SUCCESS, even as the drone is sitting on the ground with motors off.

closed 2019-05-30 by hummelslei