dji-sdk/Onboard-SDK #278: Could OSDK provide hearbeat mechanism to control A3/M600 with RC off?

magenious, 2018-03-16:
As title, we would like to control A3/M600 with remote controller off, but currently, the drone would return home and landing for it would trigger RC lost failsafe.  I recommend the OSDK could provide api to hold the control between osdk and flight controller when RC was turned off, just like APM/PIXHAWK do. 
Thanks for your reply.

rohitsantdji, 2018-03-16:
This is already available, you can enable it in DJI Assistant 2's SDK page. Check the box marked "Enable SDK Failsafe" and in the "SDK Failsafe Action" choose "Hover".

magenious, 2018-03-21:
Hi, @rohitsantdji ,  we believe that the main issue of "Hover" options was that if failsafe happen and so the drone hover in **long distance flight**, it's hard to let it land or return to home.  So in our flight setting, we are inclined to set failsafe as "RTH".
Currently, we used the following code as heartbeat between main board and flight controller:
```
	// 50Hz heartbeat signal to sustain the link between 
	// the mainboard and flight controller
	thread HeartbeatThread([&](){
		while (true){
			Sleep(20);

			Telemetry::Quaternion q = pVehicle->broadcast->getQuaternion();
			DY::Quaternion qData(q.q0, q.q1, q.q2, q.q3);
			DY::Vector3f oAttitude = DyUtils::ToEulerAngle(qData);
			float fYaw = oAttitude.z * DyUtils::RAD2DEG;	// convert rad to degree

			Control::CtrlData oCtrlData(Control::HorizontalLogic::HORIZONTAL_POSITION,
				0, 0, 0, fYaw);
			pVehicle->control->flightCtrl(oCtrlData);
			std::cout << DyUtils::GetMilliTimestamp()<<std::endl;
		}
	});
```
But I think that is a workaround, it's better that the heartbeat signal is irrelevant to flight logic.
Thanks for your reply.

rohitsantdji, 2018-04-11:
As per the title, I understand that you want the OSDK to take over control when the RC is turned off. However from the last reply it seems that you would like the aircraft to execute RTH if RC is lost by error in long distance flight. Can you help to clarify the difference between these use cases? Also, could you link to the relevant part of the PixHawk doc?

magenious, 2018-04-12:
Hi, @rohitsantdji   No,I would like the aircraft to execute RTH if sdk failsafe, that is, the aircraft would RTH if heartbeat signal is abnormal because of SDK crash or other reason. 
The heartbeat solution of Pixhawk is like this: [http://qgroundcontrol.org/mavlink/start](url) and [http://www.mavlink.org/mavlink/create_new_mavlink_message](url)
You can see that the heartbeat message is:
```
#define MAVLINK_MSG_ID_HEARTBEAT 0
 
typedef struct __mavlink_heartbeat_t
{
 uint32_t custom_mode; ///< Navigation mode bitfield, see MAV_AUTOPILOT_CUSTOM_MODE ENUM for some examples. This field is autopilot-specific.
 uint8_t type; ///< Type of the MAV (quadrotor, helicopter, etc., up to 15 types, defined in MAV_TYPE ENUM)
 uint8_t autopilot; ///< Autopilot type / class. defined in MAV_CLASS ENUM
 uint8_t base_mode; ///< System mode bitfield, see MAV_MODE_FLAGS ENUM in mavlink/include/mavlink_types.h
 uint8_t system_status; ///< System status flag, see MAV_STATUS ENUM
 uint8_t mavlink_version; ///< MAVLink version
} mavlink_heartbeat_t;
```
They adopt a protocol named mavlink and defined heartbeat message, which is send to serial comm to sustain the connection between main board and flight controller, the aircraft would trigger failsafe if the heartbeat signal connection is broken.

mithunvarma, 2018-04-27:
That’s a good option for Sdk failsafe. @rohitsantdji  is there a similar sdk failsafe option for m100 ? 
Apart from BVLOS flight missions, even when drone crashes and the uart cable gets detached or onboard computer gets affected, this option is crucial as just before crash the drone would be in sdk control mode and we cannot turn off drone with RC unless we switch back to P mode.
One more option is to detect crash with imu/esc combination and turn off the motors so further damage is restricted. During. Recent crash I observed you turned off one of the motor and not all of them even when drone is on ground. Like turning off a single motor and trying to restarting it in mid air is useful like when drone hits a mesh and props stop, but it would be great if you could stop the motors when drone is near to ground.
Thanks

magenious, 2018-05-08:
@rohitsantdji Any updates? Thanks for your help.

closed 2019-05-30 by hummelslei