dji-sdk/Onboard-SDK #279: M210 Advanced Sensing cannot take off, USB error.

jonathan7, 2018-03-18:
When the Advanced Sensing usb cable is connected to the M210, the M210 will not allow motors to start. 

Steps to reproduce: 

- Install the Onboard SDK ROS with advanced sensing enabled onto an onboard computer (in my case, an [UpSquared](http://www.up-board.org/upsquared/) running ubuntu 16.04)
- Connect and turn on the onboard computer
- Attempt to take off, either with an RC controller or the equivalent ROS API. 
- Failure, with errors:

With a manual RC controller, nothing happens in any mode, but if the RC is connected to a CrystalSky tablet the following toast message appears upon attempting to fly: 
`Error Code (code:200). Failed to take off. Check the USB connection with aircraft. Contact DJI Support if this error persists after restarting. `

With the onboard computer over a serial connection: 
```
$ rosservice call /dji_sdk/drone_task_control "task: 4" 
result: False
cmd_set: 1
cmd_id: 1
ack_data: 200
```
Additionally the following message appears in the ROS thread:
```
STATUS/1 @ getErrorCodeMessage, L644: droneTaskCallback
STATUS/1 @ getCommonErrorCodeMessage, L714: UNKNOWN_ACK_ERROR_CODE
```

Without Advanced Sensing, there are no issues with flight (through manual RC or sdk control). 
With Advanced Sensing, there are no issues with the camera APIs; the services and topics for stereo_240p and stereo_vga work fine. But, the drone does not take off. 

The RC still works even when Advanced Sensing is enabled, according to these steps: 
1. Turn on drone, start motors using RC controller
2. Turn on onboard computer, launch DJI SDK in ROS, subscribe to camera feeds
In this case, the RC still controls the drone, but upon the motors stopping the motors will not restart until the usb is disconnected or the onboard computer is powered off. 

It appears there is some safety feature preventing arming when a USB cable is connected, but the error messages do not suggest a fix.

amenonDJI, 2018-03-19:
@jonathan7 I would recommend taking a look at [M210 Checklist page](https://developer.dji.com/onboard-sdk/documentation/M210-Docs/main.html) and specifically [M210 Real World Test](https://developer.dji.com/onboard-sdk/documentation/M210-Docs/real-world-test-checklist.html) page to solve the problem of USB connected flight.

jonathan7, 2018-03-19:
Thanks for the quick response, looks like 
`./M210ConfigTool --usb-port /dev/ttyACM0 --config-file UserConfig.txt --usb-connected-flight on`
is my issue, specifically the fact that it doesn't persist between reboots. 

Is error code 200 specific to this safety feature? Should there be a more detailed error message in Onboard SDK ROS?

amenonDJI, 2018-03-19:
@jonathan7 We can look at adding a more meaningful error message for this issue.

closed 2018-03-19 by jonathan7