dji-sdk/Onboard-SDK #284: Drone version not obtained with M100 ubuntu 16.04

Gerjen100, 2018-03-29:
I have been trying to get any of the sample programs to work with the Matrice M100 but whenever i try to to run the sample program it runs into a communication error where the drone version cannot be obtained. It worked the first time and after that the output is always the same error.

[image]

I am using a usb to TTL cable so in combination with an intel NUC. The firmware on the matrice is up to date. I have tried all of the suggestions the program makes to solve this issue but it does not seem to work. Can anyone tell me a possible source for this problem.

Kind Regards Gerjen

rohitsantdji, 2018-03-29:
Can you post a screenshot of the DJI Assistant 2 SDK page, as well as a picture of your UART connections?

Gerjen100, 2018-04-03:
Here are my DJI assistant settings  : 

[image]

And here is a picture of my UART connections
[image]

And here is my Userconfig.txt
[image]

rohitsantdji, 2018-04-03:
Do you happen to know the model of the USB-Serial converter chip inside your cable? Also, just to confirm, did you try inverting the Rx/Tx pins?

Gerjen100, 2018-04-10:
The issue randomly disappeared over night. We're not sure why but it is working now.

closed 2018-04-10 by Gerjen100