dji-sdk/Onboard-SDK #287: M210 Configuration Tool not disabling the Advance sensing feature

qboticslabs, 2018-04-04:
Hello

We are working with M 210 drone with TX2 onboard computer with the ROS interface. We have tried to disable the advance sensing feature using the following tutorial

[https://developer.dji.com/onboard-sdk/documentation/M210-Docs/simulation-checklist.html](url)

The tool is running but not disabling the Advance sensing. Ubuntu version is 16.04 and ROS is Kinetic.

zliDJI, 2018-04-04:
You need to re-build and install the osdk core **without** the -DADVANCED_SENSING=ON flag.

qboticslabs, 2018-04-05:
Hi

Thanks for the reply. We need advance sensing and need to do simulation in DJI Assistant. We are looking for solutions in order to disable Advance sensing during simulation. The simulation checklist is mentioning like that only.

So if M210Config tool can disable advance sensing, we can start the ros interface and work with simulation. After simulation, we can switch it back to advance sensing. Is this possible using this config tool ?

SamuelWangDJI, 2018-04-07:
M210ConfigTool cannot disable Advanced Sensing. It's a tool to turn on/off simulation programmatically.

If you don't need the simulation visualizer inside DJI Assistant, use this M210ConfigTool to turn simulation on and do your application.

If you need simulation visualizer, you can either 
1. Use DJI Assistant, but set **enable_advanced_sensing** boolean to false in [here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/f3b136eb769e23f9128ce77babc9d7584659b707/dji_sdk/src/modules/dji_sdk_node.cpp#L85-L106). So the vehicle constructor won't complain about not finding USB.

2. Use DJI Assistant, un-define the **ADVANCED_SENSING** preprocessor macro in [here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/f3b136eb769e23f9128ce77babc9d7584659b707/dji_sdk/src/modules/dji_sdk_node.cpp#L85-L106). 

3. Use your own simulation such as Gazebo, use M210ConfigTool to turn the simulation on and pipe the telemetry topics to your simulation.

Option 1 & 2 save you the work on simulation but you cannot use Advanced Sensing at the same time. Option 3 gives you the ability to use both together.

closed 2019-05-31 by hummelslei