dji-sdk/Onboard-SDK #288: Intel Nuc Onboard computer set up

Alikhajehei94, 2018-04-06:
can anyone help me with the set up of the Intel Nuc with M210. I have created a program and was able to run it on the simulation, now I want to test it with the actual drone. What parts and equipment do I need?

I'm assuming the set up would be as follows:
Intel Nuc would somehow be mounted on the M210.
UART-USB would be connected
Wireless mouse and keyboard.
What type of power cable adaptor do I need to be able to use the drone battery as my power source for Intel nuc?
How would I connect to a monitor? Do I need some sort of wireless monitor connection?

Thanks,
Ali

SamuelWangDJI, 2018-04-07:
Power:
The XT30 port from M210 provides unregulated voltage from 18 to 26 V with a current of 2 Amp. Check the [manual](https://dl.djicdn.com/downloads/M200/20180213/Matrice+210_210+RTK+User+Manual_EN(1).pdf) for detail (page 41). 
Please check the power spec of your Intel Nuc and purchase the power cable adaptor accordingly.

Monitor:
SSH via Wifi would be a convenient option.

Asylbeck, 2019-04-05:
The power output from the M210 is not enough for the NUC (mine is i5) - as I look at the dc-dc converter (which also shows current draw) the shutdown happens as the current increases momentarily.  Mine shutdowns after it tries to boot into ubuntu.  I am now considering a small 6S battery, which is very sad.  S900 - would allow to draw power from the battery - directly.  Just curious if anyone has solved this issue some other way - without adding a small 6S lipo :))  Odroid XU4 just doesn't work - atrmv7 advanced sensing isn't supported (the library doesn't match the headers), plus the USB3 on odroid has a bandwidth of usb 2.

hummelslei, 2019-05-31:
Close this issue now. If you still have more issues related to DJI SDK, please help to fill in this form: 
https://formcrafts.com/a/dji-developer-feedback-en to report them to DJI Developer Support. Thanks for your understanding and cooperation.

closed 2019-05-31 by hummelslei

ShahAkibSarwar, 2019-06-27:
Were you able to solve this problem? I am also having a similar issue. I am trying to power an Intel Nuc (requires 19V power supply) from my DJI Matrice 100. I tried using a voltage regulator to get 19V from the XT30 port on the Matrice and it worked fine at first. However, now the NUC shuts down abruptly when it is about to boot into Ubuntu.