dji-sdk/Onboard-SDK #290: M210 Drone: Unknown error(code:99) can't takeoff

qboticslabs, 2018-04-09:
Hi

We got a weird error today when we connected the USB of M 210 to our Onboard TX2 computer, it shows the following error

"Unknown error(code:99) can't take off. Please let me know the possible solutions for this. I am attaching the screenshot 

The USB Mode of M 210 was in PC mode. Following is the link to the screenshot.

https://files.slack.com/files-pri/T035NV0V7-FA48LKLBH/img_20180408_150126.jpg

zliDJI, 2018-04-09:
@qboticslabs The link to your image requires sign up, could you post image directly here or through some other service?

qboticslabs, 2018-04-09:
Here is the image.

We are trying to follow the real world checklist link, but M210Config tool is not making any impact. We are using Arm8 version of M210Configtool

https://developer.dji.com/onboard-sdk/documentation/M210-Docs/real-world-test-checklist.html

[image]

qboticslabs, 2018-04-09:
This is the error getting on Ipad, when i connected to the controller

[image]

SamuelWangDJI, 2018-04-09:
Can you provide the output of ConfigTool?

qboticslabs, 2018-04-09:
Here is the output. I have changed the USB name to /dev/ttyM210, there is no problem with that. We have tested

[image]

SamuelWangDJI, 2018-04-09:
I didn't see anything about it successfully set USB connected flight.

**STATUS/1 @ processReceivedData, L96: Successfully set USB connected flight!**

qboticslabs, 2018-04-09:
Yea, its not showing even for setting simulation!

zliDJI, 2018-04-09:
Please try to use the x86 version of config tool. Also, in the command,
./M210ConfigTool --usb-port /dev/ttyACM0 --config-file UserConfig.txt --power-supply on

don't change the ttyACM0, because it's another port through the USB and is different from the api port.

qboticslabs, 2018-04-09:
@ziyangli  x86 version is not working in TX2. We changed the USB port and verified the ROS driver and everything was working fine. Also this problem was there before changing too.

I am getting the same result for power-supply on. Same message as I posted for usb problem. If you can publish the source code fo M210Configtool, it will be great

zliDJI, 2018-04-09:
Take an x86 PC and run config tool from there.  Probably your TX2 doesn't have ACM driver installed.

qboticslabs, 2018-04-09:
@zliDJI i will try from x86 machine. 

We already run the ROS driver using the same port. Is that different from this tool? Is it need separate driver other than running the examples code and ROS driver? Just for clearing my doubts.

zliDJI, 2018-04-09:
Yes. This tool will use both the API serial port (ttyUSB0) and another port (exposed as ACM0 through usb). You'll need to have ACM kernel driver support. We found that on TX2 with connect tech carrier board, the official connect tech kernel does not include the ACM driver and you need to rebuild the kernel by your self.  On the nvidia TX2 dev board (the big one released by nvidia), the ACM driver is included, so the armv8 configtool will run on that board.

That's why I suggested that you try the x86 version, hopefully you won't see the error. Then we can open another thread on how to build ACM driver for TX2+connect tech carrier board.

qboticslabs, 2018-04-09:
Cool. Got the problem. Thanks for clarifying. I will install ACM driver and get back to you.

qboticslabs, 2018-04-09:
It worked. I have re-compiled the kernel, enabled the CDC module. Thanks for the help :)

closed 2018-04-09 by qboticslabs