dji-sdk/Onboard-SDK #292: Faced issue when running Qt sample in Windows 10

imkhairulikhwan, 2018-04-12:
Hi and good day, 

I currently connect M210 using USB to PC Windows 10. However, when I was running the sample in Qt for Windows 10, I got following message 

"Drone version not obtained. Please do not proceed."

Although I have already enabled the API in DJI Assistant 2 and set the correct Baudrate.

Thank you!

rohitsantdji, 2018-04-17:
>I currently connect M210 using USB to PC Windows 10 

Are you connecting from the UART on the M210 to the USB on your PC with a USB-UART cable as per [this page](https://developer.dji.com/onboard-sdk/documentation/development-workflow/hardware-setup.html)?

imkhairulikhwan, 2018-04-19:
Hi @rohitsantdji, I have just received USB-TTL cable and connected it to M210 through Rx, Tx, and Ground cable according to the diagram. However, I'm still receiving the same error message in both Windows and Linux, with respective samples in Qt and Linux C++. 

Am I missing something like driver or additional step? Just FYI, I am able to access DJI Assistant to turn on the SDK in the setting.
 
[image]

[image]

RavenTKF, 2018-04-19:
Same problem

imkhairulikhwan, 2018-04-23:
@rohitsantdji Thank you! Sorry for the late reply. Actually the connection initially was based on the color code given from [Amazon](https://www.amazon.com/ADAFRUIT-INDUSTRIES-954-SERIAL-RASPBERRY/dp/B00DJUHGHI/ref=sr_1_5?s=electronics&ie=UTF8&qid=1466208644&sr=1-5&keywords=usb+to+ttl), to fit according to this diagram:

https://developer.dji.com/onboard-sdk/documentation/development-workflow/hardware-setup.html

Now I am not receiving that message anymore. But we got new message "Virtual RC is not supported". I have tried to connect the drone with the actual RC as well. The drone has the green signal and the remote show that it's connected though.

Should I run different sample or something? 

Thank you! 

[image]

rohitsantdji, 2018-04-23:
That's not a problem, it just tells you that Virtual RC as a feature is not supported on the M210. Only the M100 supports it, and you don't need to be worried about it since it is not required for any SDK-related operation. 

Additionally, your NEW_DEVICE_ERROR means that the activation credentials you tried are incorrect.

I am closing this issue since your original problem has been resolved. If you have other problems that continue and are not mentioned in the [documentation](https://developer.dji.com/onboard-sdk/documentation/introduction/homepage.html) or [API Reference](https://developer.dji.com/onboard-api-reference/index.html), please open a new issue. Thanks!

closed 2018-04-23 by rohitsantdji

imkhairulikhwan, 2018-04-24:
Noted on that, indeed the original problem has been resolved. 

Thank you for the superb assistance, really appreciate it. 

Have a nice day!