dji-sdk/Onboard-SDK #293: Bug: M100 broadcast fails after some time

hdiethelm, 2018-04-13:
Hello,

I have a problem on the M100 drone. After some time, I don’t get broadcast flight data (CMD Set 0x02 Push Data CMD Set / CMD ID 0x00 Flight Data) any more. Mostly, it happens after 4 to 5 takeoffs, but it also happened one time on ground and sometimes right after the first take off.

When my own software based on the DJI OSDK is restarted, everything looks normal but there is still no broadcast data and all flight data values are zero. The only way is to restart the complete drone, which is quite annoying especially if you have some members of the press around and you and you should demonstrate your drone.

I tracked the problem down to dji_vehicle.cpp Vehicle::PushDataHandler(void* eventData) and added debug output on each type of boradcast. In the normal condition, OpenProtocol::CMDSet::Broadcast::broadcast and OpenProtocol::CMDSet::Broadcast::mission are sent regularly:
`
Read App ID
User Configuration read successfully. 

STATUS/1 @ linux_serial_device.cpp:init:37: Attempting to open device /dev/ttyTHS1 with baudrate 115200...

STATUS/1 @ linux_serial_device.cpp:init:47: ...Serial started successfully.

DEBUG/1 @ dji_open_protocol.cpp:allocSession:127: Allocation size 19
DEBUG/1 @ dji_open_protocol.cpp:encrypt:1152: data len: 19

DEBUG/1 @ dji_open_protocol.cpp:sendInterface:408: Sending session 2

DEBUG/1 @ dji_vehicle.cpp:processReceivedData:890: Dispatcher identified as push data
STATUS/1 @ dji_vehicle.cpp:PushDataHandler:1310: broadcast
DEBUG/1 @ dji_open_protocol.cpp:appHandler:741: Recv Session 2 ACK

DEBUG/1 @ dji_open_protocol.cpp:freeSession:167: session id 2

DEBUG/1 @ dji_vehicle.cpp:processReceivedData:880: Dispatcher identified as blocking call

STATUS/1 @ dji_vehicle.cpp:parseDroneVersionInfo:568: Device Serial No. = 041DE21247

STATUS/1 @ dji_vehicle.cpp:parseDroneVersionInfo:570: Hardware = M100

STATUS/1 @ dji_vehicle.cpp:parseDroneVersionInfo:571: Firmware = 3.1.10.0

STATUS/1 @ dji_vehicle.cpp:parseDroneVersionInfo:574: Version CRC = 0xA6453AAC

STATUS/1 @ dji_vehicle.cpp:functionalSetUp:126: Subscriber not supported!

STATUS/1 @ dji_vehicle.cpp:functionalSetUp:164: MFIO not supported!

ERROR/1 @ dji_vehicle.cpp:functionalSetUp:182: Hardware Sync not supported!

STATUS/1 @ dji_vehicle.cpp:activate:951: version 0x3010A00

DEBUG/1 @ dji_vehicle.cpp:activate:952: 00000000000000000000000000000000
DEBUG/1 @ dji_open_protocol.cpp:allocSession:127: Allocation size 62
DEBUG/1 @ dji_open_protocol.cpp:encrypt:1152: data len: 62

DEBUG/1 @ dji_open_protocol.cpp:sendInterface:408: Sending session 2

DEBUG/1 @ dji_vehicle.cpp:processReceivedData:890: Dispatcher identified as push data
STATUS/1 @ dji_vehicle.cpp:PushDataHandler:1310: broadcast
STATUS/1 @ dji_broadcast.cpp:unpackCallback:23: unpackCallback!

DEBUG/1 @ dji_vehicle.cpp:processReceivedData:890: Dispatcher identified as push data
STATUS/1 @ dji_vehicle.cpp:PushDataHandler:1310: broadcast
STATUS/1 @ dji_broadcast.cpp:unpackCallback:23: unpackCallback!

DEBUG/1 @ dji_open_protocol.cpp:appHandler:741: Recv Session 2 ACK

DEBUG/1 @ dji_open_protocol.cpp:freeSession:167: session id 2

DEBUG/1 @ dji_vehicle.cpp:processReceivedData:880: Dispatcher identified as blocking call

STATUS/1 @ dji_vehicle.cpp:activate:964: Activation successful

DEBUG/1 @ dji_open_protocol.cpp:allocSession:127: Allocation size 19
DEBUG/1 @ dji_open_protocol.cpp:encrypt:1152: data len: 19

DEBUG/1 @ dji_open_protocol.cpp:sendInterface:408: Sending session 2
[... 114 lines cut ...]
DEBUG/1 @ dji_vehicle.cpp:processReceivedData:890: Dispatcher identified as push data
STATUS/1 @ dji_vehicle.cpp:PushDataHandler:1352: mission
DEBUG/1 @ dji_vehicle.cpp:processReceivedData:890: Dispatcher identified as push data
STATUS/1 @ dji_vehicle.cpp:PushDataHandler:1352: mission
DEBUG/1 @ dji_vehicle.cpp:processReceivedData:890: Dispatcher identified as push data
STATUS/1 @ dji_vehicle.cpp:PushDataHandler:1352: mission
DEBUG/1 @ dji_vehicle.cpp:processReceivedData:890: Dispatcher identified as push data
STATUS/1 @ dji_vehicle.cpp:PushDataHandler:1352: mission
`
Is this a known bug? Can you do something about?

My configuration is:
M100 Drone Firmware 1.3.1.00
Manifold connected to M100 over UART2
DJI OSDK Version 3.3.1

I tested allready reducing the baud rate to 115200 but this did not help. It seams it is a bug inside the flight controller because restarting the manifold or the OSDK based software does not help.

The config of the M100 is:
[image]
[image]
[image]

Regards, 
Hannes

rohitsantdji, 2018-04-17:
Reducing the baud rate and continuing to receive all data at 50Hz as well as ground station data (the mission push data) will probably not help with this issue. Can you do a test where you turn off the ground station data, and increase the baud rate to 230400 or 921600, and set some of the data to 0Hz instead of 50 Hz? Let us know if this issue still shows up.

tompe17, 2018-04-29:
I think we are seeing something similar on some of our M100. We use the ROS SDK and what happens sometimes is that we loose flight status data and the rc data (that was the ones I noticed). But we still get the gps position data and the attitude. The only way to get it back is to restart the vehicle.

L4ncelot, 2019-02-26:
I'm having similar issue with my M600. On raspberry pi after some time data from M600 stops comming. On my pc this happens as well with only difference in time this happens. On rpi the problem happens within 1 minute, on my pc within few minutes.

closed 2019-05-31 by hummelslei