dji-sdk/Onboard-SDK #299: Multiple Devices connected to UART

JulV94, 2018-04-17:
Hi,
I have a Matrice 100 drone and I need to connect 2 devices to it, fortunately there is 2 UART_CAN2 ports available on the drone. The first device is a SabreLite and the other is a Raspberry Pi 3 B+. When I connect one of the 2 it works perfectly but when I connect the 2 at the same time, one doesn't work. I got this error on the raspberry pi side :

```
Read App ID
User Configuration read successfully.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 115200...

STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1339: Drone version not obtained! Please do not proceed.
Possible reasons:
    Serial port connection:
        * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
        * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
        * TX and RX pins are inverted.
        * Serial port is occupied by another program.
        * Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

ERRORLOG/1 @ activate, L1228: Unable to initialize some vehicle components!
STATUS/1 @ getErrorCodeMessage, L644: initVehicle
STATUS/1 @ getCommonErrorCodeMessage, L710: ACK_NO_RESPONSE_ERROR
Vehicle not initialized
```

Everything listed in the possible reason cannot be the reason sinnce the code is working as soon as I remove the SabreLite. And for the Serial ports used, I've used each UART_CAN2 ports so I assume they are different ports. The 2 devices don't use the same version of the OSDK, maybe that's the reason, is the UART protocol used by the drone dependent of the OSDK version?

rohitsantdji, 2018-04-17:
@AutisticNinja is right, there is a single UART within the M100 that is broken out into two ports for ease of access, not for connecting multiple devices simultaneously. In fact, trying to send data from two computers will cause both to behave erratically.

closed 2018-04-17 by rohitsantdji

JulV94, 2018-04-20:
Thank you for the fast responses.

I forked the code and managed to connect 2 devices on the drone by bypassing the acknowledgement part. Of course only 1 device is sending commands to the drone, the other is only listening to the broadcast (RX of the drone is disconnected). I put the link here if it can help someone in the same situation as me [https://github.com/JulV94/Onboard-SDK](url)

pavankumarbn, 2018-05-01:
Hello,

I have been trying to connect Matrice 100 to onboard Intel NUC. Using 3.6 SDK Version but facing following error. Can you please give any suggestions how to fix this issue?

I  directly wired  TTL end of USB-TTL to UART CAN 2' end. Is it a right way to connect these two? Is there any other way to connect.

Your suggestions will help me a lot to fix this issue. 

Read App ID
User Configuration read successfully.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1339: Drone version not obtained! Please do not proceed.
Possible reasons:
    Serial port connection:
        * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
        * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
        * TX and RX pins are inverted.
        * Serial port is occupied by another program.
        * Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

ERRORLOG/1 @ activate, L1228: Unable to initialize some vehicle components!
STATUS/1 @ getErrorCodeMessage, L644: initVehicle
STATUS/1 @ getCommonErrorCodeMessage, L710: ACK_NO_RESPONSE_ERROR
Vehicle not initialized, exiting.

imkhairulikhwan, 2018-05-01:
Hi, 

you might want to try to switch the cable's connection as I faced similar issue before:

https://github.com/dji-sdk/Onboard-SDK/issues/292

Hope it helps.

pavankumarbn, 2018-05-02:
@imkhairulikhwan Thank you very much:) It is working now.

imkhairulikhwan, 2018-05-02:
Awesome!