dji-sdk/Onboard-SDK #301: Recommended data for flight control

pingualoty, 2018-04-17:
We have an M600 with RTK and are using a heavily modified moveByPositionOffset to control the drone flight along lines. We want to use the RTK for controlling position during flight to keep it on the lines, however we are concerned that it appears the raw RTK data is only output at 5Hz.
Even though DJI Assistant has the option to output much faster, the same applies to the raw GPS data, it is also 5Hz.

1. So should be using the raw RTK data to control flight?
2. Or should we use the data from getGlobalPosition which is updated much faster and seems to not be raw GPS or RTK and we presume it is augmented with IMU data etc? 
3. Is RTK data incorporated into getGlobalPosition?

If you could answer those quick questions it would be really helpful!

rohitsantdji, 2018-04-17:
That's a good observation, that is indeed correct. The raw update rate of the GPS and the RTK is 5Hz. 

1. No, that's probably not a good idea since this data has no real outlier rejection or filtering etc.
2. This is your best choice, it is the full state output that the aircraft itself is using for position control. It is the output of the internal sensor fusion and will give you all the benefits of fusion, redundancy, filtering etc.
3. Yes, the sensor fusion will take RTK data into account when it is available.

rohitsantdji, 2018-04-19:
Thank you for the detailed testing and explanation. We are looking into this issue internally. I hope to get back to you soon about this.

rohitsant, 2018-06-22:
Hey @pingualoty - sorry for the delayed response. Here are some details on the RTK incorporation into our data streams:

1. I was mistaken in saying that the getGlobalPosition data incorporates RTK; it is in fact just GPS data. Your tests also confirm this beyond any doubts. That API is part of our old, Broadcast-style data streams

2. In our new, Subscription-style data streams, as of OSDK 3.6 we also don't have a data topic that incorporates RTK position into a fused position output. However, **we will be offering a position topic with RTK fusion** with the  release of OSDK 3.7 in Q3 (hopefully quite soon). Aircraft will need firmware updates to support the new data streams available with OSDK 3.7 so you might have to wait a little bit before your M600 becomes fully compatible. 

3. Currently, our **velocity** topic offered through the subscription stream **does take RTK** into account while doing its fusion and is available at up to 200Hz; so that could be a (slightly convoluted) workaround until 3.7 support is available.

Thanks for your patience!

mabrowning, 2018-08-15:
@rohitsant @rohitsantdji This is a rather critical feature for us, and I'm looking through the onboard 3.7 documentation to find the new `position topic with RTK fusion`, but all I see is that the documentation for `TOPIC_GPS_FUSED` no longer mentions RTK. What am I missing?

Thanks!

pingualoty, 2018-08-15:
@mabrowning There is 3.7 documentation, where is that located?

mabrowning, 2018-08-15:
The doxygen comments in [dji_telemetry_doc.hpp](https://github.com/dji-sdk/Onboard-SDK/blob/3.7/osdk-core/api/inc/dji_telemetry_doc.hpp)

rohitsantdji, 2018-08-16:
@mabrowning Glad you find the new API docs more useful! They should also be up on the website soon.

The new position topic with RTK fusion is `TOPIC_VO_POSITION` - it doesn't mention RTK in `dji_telemetry_doc.hpp`, that's an oversight on our part and we've fixed it in the version that goes on to the website. The naming has VO front and center because that's the primary source of positioning information but it will use GPS and RTK (in fact every sensor that can contribute to position).

In the future as well, we'll be updating the website docs more frequently rather than push out new point releases to the repo, so you can treat those as a source of truth.

mabrowning, 2018-09-04:
@rohitsantdji Ok, I've done extensive testing. This new topic is good; I want to be able to use it, but it has many caveats and problems. The fact that x,y are not necessarily N,E is concerning, but we'll always be operating in a clear sky with a good compass, so I'm just assuming they are N,E aligned...

I have had cases where xHealth and yHealth transitioned from 0->1, but on the next read, the values jumped by `(-3411260.0, 10719108.0)` (3% the distance to the moon...). I had been using xHealth and yHealth to determine when to grab the reference point, but now I'm using it *and* some discontinuity logic. Fine.

Also, it appears that the Z value is never "good" (zHealth=0 always), and that value drifts all over the place: (green is `TOPIC_POSITION_VO`, blue is `TOPIC_RTK_POSITION` (the naming inconsistency is also infuriating), and red is the `TOPIC_GPS_FUSED`)
[image]

`TOPIC_RTK_POSITION` is extremely stable:
[image]

But for that same duration, the `TOPIC_POSITION_VO` floats +- 2m (admittedly, zHealth=0):
[image]

I have Visual Positioning turned *off*, because it was even worse with it on. I just want RTK + IMU fusion and nothing to do with the visual odometry sensors. At this point, I'm going to need to implement my own linear kalman filtering based on `TOPIC_VELOCITY` + `TOPIC_RTK_POSITION` &`TOPIC_QUATERNION` (because is Antenna 1, not drone body center).

It seems crazy to me that you can't integrate RTK into the sensor fusion for `TOPIC_GPS_FUSED`... what is the drone using internally for position keeping?

rohitsantdji, 2018-09-10:
@mabrowning Thanks for your tests. The 2m noise on Z does not seem normal - can you upload your flight logs (DAT files) on a cloud server and send us a link at dev@dji.com? We would like to have some of our flight control engineers examine them. Please reference this issue (#301) when you submit your logs.

closed 2019-05-31 by hummelslei