dji-sdk/Onboard-SDK #302: Advanced Sensing problem

a137755952, 2018-04-18:
when I configure the dji-core with cmake .. -DADVANCED_SENSING=ON and then I type the command roslaunch dji_sdk sdk.launch. The terminal log shows that:
STATUS/1 @ init, L49: Looking for USB device...
STATUS/1 @ init, L65: Found 5 USB devices, identifying DJI device...
ERRORLOG/1 @ init, L89: Did not find any DJI USB device, please double-check your connection.
ERRORLOG/1 @ init, L91: Some helpful commands are 'lsusb' and 'dmesg'.
STATUS/1 @ activate, L1239: version 0x3022C0B
[dji_sdk-2] process has died [pid 18570, exit code -11, cmd /home/nuc-m210/Onboard-SDK-ROS-3.6/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/nuc-m210/.ros/log/297cd4d8-42a3-11e8-bf0d-a0c5895531c4/dji_sdk-2.log].
log file: /home/nuc-m210/.ros/log/297cd4d8-42a3-11e8-bf0d-a0c5895531c4/dji_sdk-2*.log

while the problem is that If I configure the dji-core with cmake.. only, it means that I will not use the Advanced Sensing features for the M210, all goes well.

Can anyone help me with this issue?

SamuelWangDJI, 2018-04-18:
If you intend to use advanced sensing, plug in both the UART and the USB ports.
If you intend to install advanced sensing in your system but not use it, set **enable_advanced_sensing** boolean to false in [here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/f3b136eb769e23f9128ce77babc9d7584659b707/dji_sdk/src/modules/dji_sdk_node.cpp#L85-L106).

a137755952, 2018-04-20:
@SamuelWangDJI thanks for answering my question, the usb ports is connected to the simulator and UART ports is connected to my onboard-computer. So whether the advanced sensing can support simulation by computer or not?

SamuelWangDJI, 2018-04-20:
Advanced Sensing uses USB to communicate with the aircraft. You have to connect both USB and UART to your onboard computer to use Advanced Sensing. You won't be able to use DJI Assistant 2 for the visualization of the simulation at the same time. Since the simulation takes place inside the flight controller, we provide a [config tool](https://github.com/dji-sdk/Onboard-SDK/tree/3.6/utility/bin) to turn it on/off.

Please read the [documentation](https://developer.dji.com/onboard-sdk/documentation/M210-Docs/main.html) for detailed infomation.

closed 2018-04-25 by rohitsantdji

Hepan02, 2022-07-21:
Hi. I have the same problem in Onboard-SDK. How I can disable AdvancedSensing in Oboard-SDK?
I do not understand why one of my program (telemetry sample) works with information: 
_USB is not plugged or initialized successfully. Advacned-Sensing will not run._
But the same code i another program give me an error:
_[536495.401]STATUS/1 @ init, L65: Found 6 USB devices, identifying DJI device...
[536495.401]ERRORLOG/1 @ init, L89: Did not find any DJI USB device, please double-check your connection._
Can you help me with disabling **AdvanceSensing**?