dji-sdk/Onboard-SDK #303: onboard sdk 中的waypoints mission中的waypoint 数目问题

zhongzhaoji, 2018-04-18:
使用waypoints mission，任务点数小于9可以正常执行任务，一旦上传任务点数大于9，任务无法执行，并提示waypoints mission point not enough
[image]

xzq123, 2018-04-18:
老哥可不可以讲下你waypoint怎么搞的？我直接在mission sample 里面把generateWaypointsPolygon这个函数算gps那里改了，然后用一个数组记录GPS点，但是这样搞出来总是出现waypoint_mission_points_too_far

zhongzhaoji, 2018-04-18:
你用的是qt sample？例程跑通没有？

zhongzhaoji, 2018-04-18:
[image]

xzq123, 2018-04-18:
我用的Linux例程跑通了

zhongzhaoji, 2018-04-18:
你的两个gps点是不是相隔太远了

cynapoleon, 2018-04-23:
大佬，我也问个问题哈，
我用的是QT 3.6 的 sample 程序，连接的是N3飞控，传三个航点就提示 waypoint not enough.
能大致说下应该通过什么步骤上传航点么,那些数据是必须的。

zhongzhaoji, 2018-04-25:
想问下，你三个航点是一样的吗？然后如果不是一样，可以尝试在load all point的那个槽函数里的for循环加一段延时函数尝试下，大概7ms @cynapoleon

closed 2019-05-31 by hummelslei