dji-sdk/Onboard-SDK #307: Write exif data into the image

imkhairulikhwan, 2018-04-24:
Hi and good day, 

I am using M210 to access OnBoard SDK using Qt sample.

I tried to look around the forum and the documentation, and it seems not raised yet. I would like to use OnBoard SDK to write exif data e.g. distance from the object into the image. 

Hence, I need to access the image from SD card and write the exif data into the image. Or maybe override taking photo function? 

However, from this link: https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-camera-and-gimbal.html

it seems I am not able to access Media Manager through the sdk. Is there any way? Besides using Mobile SDK. 

Thank you!

rohitsantdji, 2018-04-25:
Unfortunately at this time we don't support managing/manipulating images through the Onboard SDK.

closed 2018-04-25 by rohitsantdji

imkhairulikhwan, 2018-04-26:
Noted on that, thank you for the confirmation.

OlegJakushkin, 2025-01-27:
It is now 2025, is this feature supported?