dji-sdk/Onboard-SDK #309: Remote Controller Lost, mission was suspended not like described in manual.

siyuluuu, 2018-04-24:
A3 version: 1.7.1.5
OSDK: 3.4
Enable SDK Failsafe Action: ticked.
SDK Failsafe Action Settings: Hover

Mission was uploaded through onboard SDK before takeoff and executed, the drone hovered when RC was switched off. The desired behavior is to continue the mission. This is not aligned with descriptions in
https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-control-authority.html

What else should be done to continue the mission?

rohitsantdji, 2018-04-25:
Did you change your [RC lost action](https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1WayPointInitSettings.html#aa46da1cfda3af725541e1b897c8a998f) to 1?

siyuluuu, 2018-04-26:
found another process was sending velocity at the same time. Deleted it and the mission is completed successfully.
Thanks.

closed 2018-04-26 by siyuluuu