dji-sdk/Onboard-SDK #311: Outer loop PID parameters in simulator

rocklinsuv, 2018-04-27:
Hi there,

Recently I've been trying to implement a PID outer loop controller for my M100. I used HORI_ATTI_TILT_ANG for horizontal control, and VERT_POS as well as YAW_RATE for z/yaw control. I would like to know if you guys have tried similar stuff before? If so, what would be the possible P/I/D values for such a controller? What I got was something like P(around 0.2)/I(less than 0.02)/D(around 0.2) in the simulator. But it shouldn't be such small values according to my experience.
If anyone have done similar things before, it would be very much appreicated if you can share your experience with me.

Best Regards,

mithunvarma, 2018-04-27:
Hi @rocklinsuv,

I tried something similar with default horizontal ground vel and yaw rate default mode velocity control, 
Could you upload screenshot of trace in simulator or plot of your path for sample run using rqt multi plot.
What’s the freq/rate of your control loop. I tried contacting DJI Dev support team, only thing that they conveyed is you should you 50 hz control loop.
Did you try your pid control loop at different rate other than 50hz and how does your plots look like? 
Thanks,

rocklinsuv, 2018-04-27:
Hi @mithunvarma,

Thanks for your reply. I don't have access to my drone ATM. Once available I'll upload the data. I didn't try other frequencies except the default 50Hz. I believe there is a ros::sleep() somewhere in the SDK which you may modify to suit different frequencies. I'll let u know where when I get back to work, along with the controller I tuned.

rocklinsuv, 2018-05-02:
@mithunvarma Just got back from holiday.

I've done two tests in the simulator, velocity/angle based control.
In both tests, the control commands were published to topic "dji_sdk/flight_control_setpoint_generic" at 50Hz. Besides, I modified the "demo_flight_control.cpp" to let M100 take off, and travel to the first waypoint, then stop.

1. Velocity control. I set the drone to travel at a fixed speed of 1m/s.

sensor_msgs::Joy controlGeneric;
uint8_t flag = (DJISDK::VERTICAL_POSITION |
		       DJISDK::HORIZONTAL_VELOCITY |
		       DJISDK::YAW_RATE | 
		       DJISDK::HORIZONTAL_GROUND |  
		       DJISDK::STABLE_ENABLE);

  controlGeneric.axes.push_back(1);        // 1m/s for x
  controlGeneric.axes.push_back(0);       // 0m/s for y
  controlGeneric.axes.push_back(2);       // 2m for height
  controlGeneric.axes.push_back(0);       // fixed yaw
  controlGeneric.axes.push_back(flag);	//
  ctrlGenericPub.publish(controlGeneric);

The results look like this:
[image]

Note that the x axis corresponds to roll, and I convert the angle unit from radius to degree. I'm still not convinced that such a small angle (0.1 degree) will keep the drone at a constant speed in reality.

2. Angle control. That's how I implemented my outer loop PID at the very beginning. 
In this test I set the drone to travel at a fixed angle of 5 degrees.

sensor_msgs::Joy controlGeneric;
uint8_t flag = (DJISDK::VERTICAL_POSITION |
		       DJISDK::HORIZONTAL_ANGLE |
		       DJISDK::YAW_RATE | 
		       DJISDK::HORIZONTAL_GROUND |  
		       DJISDK::STABLE_ENABLE);

  controlGeneric.axes.push_back(5*deg2rad);        // 5 degs for x
  controlGeneric.axes.push_back(0);       // 0 deg for y
  controlGeneric.axes.push_back(2);       // 2m for height
  controlGeneric.axes.push_back(0);       // fixed yaw
  controlGeneric.axes.push_back(flag);	//
  ctrlGenericPub.publish(controlGeneric);

Here comes the result:
[image]

I still don't understand why the angle keeps dropping......I set it to 5 degrees!
Any thoughts? What about your experiment? If you need more info, please let me know. I can easily make more plots if necessary. 
Also it would be very much appreciated if the DJI staffs can solve my doubts.
Thanks.

InterestingWalrus, 2019-04-30:
@rocklinsuv @mithunvarma Did you guys manage to solve this? I'm trying to implement PID control for the Matrice M100 using horizontal and vertical velocity flags and I can't seem to get the drone to settle on any of the axes

closed 2019-05-31 by hummelslei