dji-sdk/Onboard-SDK #312: Issue with setup

johnnyapplesed, 2018-04-30:
I'm getting this error trying to follow the instructions in the SDK guide:

CMake Error: The source directory "C:/Users/john/Downloads/Onboard-SDK-3.6" does not appear to contain CMakeLists.txt.

imkhairulikhwan, 2018-04-30:
You need to create another directory named 'build' inside the SDK folder, as stated in the instructions:

https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html

mkdir build
cd build
cmake ..
make

johnnyapplesed, 2018-04-30:
Sorry, I should have included the line directly above:

C:\Users\john\Downloads\Onboard-SDK-3.6\build>cmake ..
CMake Error: The source directory "C:/Users/john/Downloads/Onboard-SDK-3.6" does not appear to contain CMakeLists.txt.

imkhairulikhwan, 2018-05-01:
I see, you have already created the build folder. But it seems you are using Windows OS? The cmake parameters might differ from Linux I guess, as it seems can't recognise the folder to when executing cmake.

johnnyapplesed, 2018-05-01:
Right, I'm using Windows. What steps should I take instead of the ones I've been trying?

closed 2019-05-31 by hummelslei