dji-sdk/Onboard-SDK #316: Nvidia TX2 + J120 connection with DJI N3 API port problem

maciejmatuszak, 2018-05-09:
Hi,
I am having trouble connecting Nvidia TX2 + J120 carrier board with the DJI N3 flight controller.
It works fine via USB/TTL converter but connecting directly to UART on J120 it does not work.
The ERROR:
```
... logging to /home/ubuntu/.ros/log/de29a51e-5270-11e8-8ea7-00044b8c3cf9/roslaunch-tegra-ubuntu-6698.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.8.1:38352/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1046870
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: deleted from log
 * /dji_sdk/serial_name: /dev/ttyTHS1
 * /dji_sdk/thrust_coefficient: 2.7525
 * /dji_sdk/thrust_max: 100.0
 * /dji_sdk/thrust_min: 2.0
 * /dji_sdk/thrust_offset: -18.3583
 * /dji_sdk/use_broadcast: False
 * /dji_sdk/use_gear_sw_for_authority_ctrl: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

ROS_MASTER_URI=http://192.168.8.1:11311/

process[dji_sdk-1]: started with pid [6730]

STATUS/1 @ init, L55: Attempting to open device /dev/ttyTHS1 with baudrate 921600...

STATUS/1 @ _serialOpen, L173: Open else
STATUS/1 @ init, L65: ...Serial started successfully.
[ INFO] [1525755937.089124893]: Succeeded to read from serial device

STATUS/1 @ init, L55: Attempting to open device /dev/ttyTHS1 with baudrate 921600...

STATUS/1 @ _serialOpen, L173: Open else
STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1339: Drone version not obtained! Please do not proceed.
Possible reasons:
	Serial port connection:
		* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
		* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
		* TX and RX pins are inverted.
		* Serial port is occupied by another program.
		* Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

[ERROR] [1525755943.092784910]: drone activation error
[ERROR] [1525755943.092864142]: Vehicle initialization failed

```
I am using the Onboard-SDK version labelled "3.6" and latest firmware: 
STATUS/1 @ parseDroneVersionInfo, L689: Hardware = N3
STATUS/1 @ parseDroneVersionInfo, L690: Firmware = 3.2.36.8

With exactly the same configuration just changing the device name to /dev/ttyUSB0 and connecting via UAB/TTL converter it works flawlessly. 
I spend some time already investigating the problem. The extra log line above: "STATUS/1 @ _serialOpen, L173: Open else" come from extra DSTATUS statement I put to check which version of code is build:

```
bool
LinuxSerialDevice::_serialOpen(const char* dev)
{
#ifdef __arm__
  DSTATUS("Open __arm__");
  m_serial_fd = open(dev, O_RDWR | O_NONBLOCK);
#elif __x86_64__
  DSTATUS("Open __x86_64__");
  m_serial_fd = open(dev, O_RDWR | O_NOCTTY);
#else
  DSTATUS("Open else");
  m_serial_fd = open(dev, O_RDWR | O_NOCTTY);
#endif
  if (m_serial_fd < 0)
  {
    DERROR("cannot open device %s\n", dev);
    return false;
  }
  return true;
}

```
When enabled debug logging the serial driver sends the message and there is no error:
```
DEBUG/1 @ allocSession, L132: Allocation size 19
DEBUG/1 @ encrypt, L934: data len: 19

DEBUG/1 @ sendInterface, L381: Sending session 2

DEBUG/1 @ sendData, L416: Open Protocol cmd send success
```

I tested the UART2 port itself by connecting it with the USB/TTL converter. This way I have 2 devices: /dev/ttyTHS1 /dev/ttyUSB0 I used "screen" program to send data trough the connection and no problem here, works both ways.
Finally I hooked up scope to the lines and the error is clearly visible.
The line from N3 to J120 looks clean and the measured baud rate is close to 921600. If I understand where the error is happening then the DJI SDK already received some data. So this path is clear.
Trace N3 -> TX2 (channel 1- bottom on the image):
https://photos.app.goo.gl/yg7P3WZ3kD9LRQek6

The other way is problematic. The line goes down (I assume during the transmission but the length varies ) but only half way to 1.68V
Trace TX2 -> N3 (channel 2- top on the image):
 https://photos.app.goo.gl/txgPHx4HTQeKco5D6

I tried to run the sample djiosdk-flightcontrol-sample with the same result. I tried it with sudo in case there are permissions problems with ttyTHS1 but no luck.
While the sample is running I took read of the UART config using stty. Nothing obvious to me here
```
ubuntu@tegra-ubuntu:~$ sudo stty -F /dev/ttyTHS1 -a
speed 921600 baud; rows 0; columns 0; line = 0;
intr = <undef>; quit = <undef>; erase = <undef>; kill = <undef>; eof = <undef>; eol = <undef>; eol2 = <undef>; swtch = <undef>;
start = <undef>; stop = <undef>; susp = <undef>; rprnt = <undef>; werase = <undef>; lnext = <undef>; discard = <undef>;
min = 0; time = 0;
-parenb -parodd -cmspar cs8 -hupcl -cstopb cread clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl -onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon -iexten -echo -echoe -echok -echonl -noflsh -xcase -tostop -echoprt -echoctl -echoke -flusho -extproc
ubuntu@tegra-ubuntu:~$ 
```
Any help would be appreciated!

zliDJI, 2018-05-09:
Can you reduce the baud rate to 230400 or even 115200 from DJI assistant 2 and try to run osdk with those baud rates?

maciejmatuszak, 2018-05-09:
You thinking the UART may not cope... according to [user manual of J120 section "UART 0 / UART 2 (J7) - rev 2 to 5 "](https://auvidea.com/download/manual/J120/J120_technical_reference_1.9.pdf) the chip used as level shifter is [TXB0104](http://www.ti.com/lit/ds/symlink/txb0104.pdf) which has 40Mbps limit, so it should not be a problem. I tried it anyway. Set the baud rate to 115200 restarted the N3, confirmed the baud on data transmitted from N3 but still no luck:
```
ubuntu@tegra-ubuntu:/data/git/dji_mav_ws/src/Onboard-SDK/build$ sudo ./bin/djiosdk-flightcontrol-sample ./bin/UserConfig.txt
Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyTHS1 with baudrate 115200...

STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1339: Drone version not obtained! Please do not proceed.
Possible reasons:
	Serial port connection:
		* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
		* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
		* TX and RX pins are inverted.
		* Serial port is occupied by another program.
		* Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

ERRORLOG/1 @ activate, L1228: Unable to initialize some vehicle components!
STATUS/1 @ getErrorCodeMessage, L644: initVehicle
STATUS/1 @ getCommonErrorCodeMessage, L710: ACK_NO_RESPONSE_ERROR
Vehicle not initialized, exiting.
```

Jai-GAY, 2019-03-19:
not sure will [this going to be helpful to you](https://github.com/dji-sdk/Onboard-SDK/issues/374#issue-359809692). i trace and [modify dji_vehicle.cpp](https://github.com/dji-sdk/Onboard-SDK/blob/3.7/osdk-core/api/src/dji_vehicle.cpp)

this code give the problem. by the way, still not solved yet.
if (ackPtr - startPtr > 64)
{
//Jai - start
DERROR("LinuxSetup::getDroneVersion B2 return false\n");
// Jai - end
return false;
}

closed 2019-05-31 by hummelslei