dji-sdk/Onboard-SDK #317: Nvidia TK1+Matrice 100+Onboard-SDK 3.6: "Drone version not obtained"

Avivm122, 2018-05-13:
Hi,
Im having problem with the SDK. Im using "Jetson TK1" as my onboard computer and Ive connected a uart cable to port "UART_CAN2". Ive put a USB to TTL device, YP-05.
While running the flight control sample Ive encountered multiple times with an error that says "Drone version not obtained! Please do not proceed." Ive been serching online for a solution but with no success.
[image]
My SDK is enabled, "Enable API Control" is ok.
The Baudrate is correct. 
TX and RX pins are correct.
And my user hace dialout permission.
Is there anyone who also encountered this?

SkullPirateTK, 2018-05-15:
I think only OSDK 3.4 supports M100.

menzehua, 2018-05-31:
hello,I use tx1 to communicate with M100 by FT232,and I meet the same error,have resovled it?thank you.  @Avivm122

Avivm122, 2018-05-31:
Hi menzehua,
I've solved the problem. 
I think that I started using the other UART port on the drone and also for running a sdk program you need to connect the wireless remote to the drone and the wireless remote to the dji go app so it will have network to check the app ID and key. 
As I remember that was the solution that worked for me.

menzehua, 2018-05-31:
I kept connection between the M100 and the remote,meanwhile the remote and the djigo app,
I can see data reception through two UART on M100 by using a serial port software.
my system is ubuntu14.04,and I have tried on a pc with ubuntu 16.04 on PC,but the same error happened.@ Avivm122

menzehua, 2018-05-31:
rcabg

menzehua, 2018-05-31:
@rohitsantdji

menzehua, 2018-05-31:
@rohitsantdji,can you offer some suggestions?thank you.

rohitsantdji, 2018-06-04:
Hi @menzehua some questions:

1. Is the voltage logic level correct on your FT232?
2. Can you confirm that you have tried all the suggestions the sample provides, such as adding user dialout permission etc? 
3. What serial port are you using on your TX1? Is the voltage level setting correct for this as well?

Avivm122, 2018-06-04:
I’ve fixed the problem.

closed 2018-06-04 by Avivm122