dji-sdk/Onboard-SDK #337: Error building SDK wtih Advanced Sensing.

mikeklinker129, 2018-06-18:
I am following the ROS onboard computer instructions on the DJI Developer site, and building the advanced sensing. [https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html#ros-onboard-computer](https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html#ros-onboard-computer)

```
sudo apt-get install build-essentials
<clone and cd into this repository>
mkdir build
cd build
cmake .. -DADVANCED_SENSING=ON
make djiosdk-core
sudo make install djiosdk-core
```

Below is my terminal when it fails out:

```
[ 74%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_thread_manager.cpp.o
[ 76%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_platform_manager.cpp.o
[ 79%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_log.cpp.o
[ 82%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_hard_driver.cpp.o
[ 84%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_memory.cpp.o
[ 87%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/utility/src/dji_circular_buffer.cpp.o
[ 89%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/utility/src/dji_singleton.cpp.o
[ 92%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/platform/default/src/dji_memory_default.cpp.o
[ 94%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/platform/linux/src/posix_thread_manager.cpp.o
[ 97%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/platform/linux/src/linux_serial_device.cpp.o
[100%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/platform/linux/src/posix_thread.cpp.o
Linking CXX static library ../libs/libdjiosdk-core.a
[100%] Built target djiosdk-core
uavracer@ubuntu:~/dji/Onboard-SDK/build$ sudo make install djiosdk-core 
[  9%] Built target advanced-sensing
[ 46%] Built target djiosdk-core
Scanning dependencies of target djiosdk-cameragimbal-sample
[ 47%] Building CXX object sample/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/camera_gimbal_sample.cpp.o
[ 48%] Building CXX object sample/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/main.cpp.o
[ 50%] Building CXX object sample/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/__/common/dji_linux_environment.cpp.o
[ 51%] Building CXX object sample/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/__/common/dji_linux_helpers.cpp.o
Linking CXX executable ../../../bin/djiosdk-cameragimbal-sample
/usr/bin/ld: ../../../../osdk-core/advanced-sensing-2.0.1/lib/libadvanced-sensing.a(dji_advanced_sensing.cpp.o): unrecognized relocation (0x2a) in section `.text'
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [bin/djiosdk-cameragimbal-sample] Error 1
make[1]: *** [sample/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/all] Error 2
make: *** [all] Error 2
uavracer@ubuntu:~/dji/Onboard-SDK/build$ 

```

rcabg, 2018-06-19:
Hey @mikeklinker129 

can you describe the pc you are using (architecture, so...)? It seems to be a problem with a linux depence.

mikeklinker129, 2018-06-19:
@rcabg This was from running Ubuntu 14.04 in a virtual machine. 

I just tried on my desktop that has 16.04 naively installed, and it worked (less some OpenCV dependency issues with sample/linux/advanced-sensing/stereo_vision_depth_perception_sample/)

mikeklinker129, 2018-06-20:
@imkhairulikhwan is CUDA required for the advanced sensing to work?

imkhairulikhwan, 2018-06-21:
@mikeklinker129 no it's not, it's just optional but it's good to have to accelerate computing. Also, the aim is to install and run the samples on the other board e.g. Jetson.

mikeklinker129, 2018-06-21:
@imkhairulikhwan Understood. Just trying to get the system working before I transition to a jetson.

imkhairulikhwan, 2018-06-22:
@mikeklinker129 yes, I am thinking we should enable CUDA as well so that when it would be smooth transition when setting up on another board

rohitsantdji, 2018-07-06:
@mikeklinker129 please see the updates in dji-sdk/Onboard-SDK-ROS#202 for workarounds to your CUDA linking issue.

closed 2018-07-06 by rohitsantdji