dji-sdk/Onboard-SDK #340: TOPIC_RTK_* seem have maximum frequency of 5Hz

asiron, 2018-06-22:
We're also using M600 with RTK and Onboard-SDK 3.6, but instead of the *Broadcast*, we are using the *Subscription* model.

The RTK topics: `TOPIC_RTK_POSITION, TOPIC_RTK_VELOCITY, TOPIC_RTK_YAW, TOPIC_RTK_POSITION_INFO, TOPIC_RTK_YAW_INFO` were all requested in a 10Hz package. However, the actual frequency seems to be around 5Hz, due to duplicate messages. We had the same problem in the past with the OSDK 3.2. 

*If this is intended, then why the maximum frequency is @50Hz instead of @5Hz?*

pingualoty, 2018-06-22:
As the author of #301 we have done more work on this since. RTK will only output at 5Hz from either Broadcast or Subscription, that seems to be it's maximum from a hardware point of view. I'm not entirely sure why they allow an attempt to Subscribe and Broadcast at higher speeds.

rohitsant, 2018-06-22:
Hi @asiron - the 5Hz max update from RTK is indeed a hardware limitation, and the ability to subscribe at higher frequencies is a bug. We will fix it in the next update.

Meanwhile, please take a look at the answer in #301 to look at data streams that may help you get faster updates with the same RTK precision in current and future versions.

closed 2018-07-06 by rohitsantdji