dji-sdk/Onboard-SDK #341: TOPIC_GPS_FUSED drifts even when RTK is present

asiron, 2018-06-22:
We have found similar problem to #301. 

We're also using M600 with RTK and Onboard-SDK 3.6, but instead of the *Broadcast*, we are using the *Subscription* model.

We convert the measurements from the `TOPIC_GPS_FUSED@50Hz` into a local Cartesian coordinate frame and we do the same for `TOPIC_RTK_POSITION@10Hz`. The animated figure below illustrates those measurements, where the *pink dot* are the measurements from  `TOPIC_GPS_FUSED` and the *red arrow* is the `TOPIC_RTK_POSITION`. The distance between those two is due to a slightly different local position reference and is irrelevant. The important thing is that, the UAV was hovering in the air, which can be observed on the *red arrow* (`TOPIC_RTK_POSITION`), while the *pink dot* (`TOPIC_FUSED_GPS`) is drifting considerably over time. The grid size is 10cm, so the drift is approximately 10 [cm/s].

[image]

As in #301, you recommend not to use the raw RTK measurements due to slow frequency, potential outliers, etc. However, the raw RTK seems to be much more stable than the actual GPS_FUSED measurements, which as you mentioned before is the full state estimation used for the internal control loop of the system. It also shows, that it should incorporate the RTK information if its present, but it most likely does not due to the present drift.

[image]

Lastly, all position measurements reported return code `50`, *integer narrow-lane ambiguity solution*. Is this ok, if not, which return codes are indicating proper operation of the RTK? We had clear sky, no tall buildings and line of sight between RTK and M600.

[image]

pingualoty, 2018-06-22:
As the author of #301 we have had more experience on this matter since that issue. We have found results that are exactly the same as yours, we get drifting positions from GPS Fused/getCurrentPosition() and a position that only moves in a few centimetres when using just RTK. 

The main issue as has been highlighted is the slow 5Hz update rate of RTK which can cause issues for precise applications when moving. We have also discovered after emailing DJI Support that the RTK position given is that of Antenna 1 of the RTK setup on the drone, not the centre of the drone or the mid point between the 2 RTK antennas. There is some reasonably simple maths to use the distance & angle the antenna is away from the centre of the drone to get that more usable.

To try and counter the drift of GPS and slow updates of RTK we have ourselves created a method that combines the 2. Though it would be far better if the DJI provided combined GPS/RTK/Onboard sensors was an accurate and non drifting position.

rohitsant, 2018-06-22:
Please see the updated answer in #301 to help confirm some of your suspicions, and to get some further options with existing and upcoming data streams that incorporate RTK data and provide faster update rates.

Regarding the position health flag value, `50` is the best case scenario for RTK accuracy. The `48-50` range corresponds to fixed solutions, these will be an order of magnitude better than the corresponding floating solutions.

closed 2018-07-06 by rohitsantdji