dji-sdk/Onboard-SDK #342: ADVANCED_SENSING FOR MATRICE 600

MrPatinhas, 2018-06-25:
Hi guys !

I am currently working with the Matrice 600 drone and I would like to obtain images from the camera (Gimbal) however I keep getting errors from the Advanced Sensing module regarding the activation of vehicle.

Is there a way of obtaining video stream for m600 ?

Thanks in advance,
Thiago

rcabg, 2018-06-25:
hey @MrPatinhas ,

I made a summary in #280 where I said that so far is not possible to obtain video stream with M600. One of the contributors (@rohitsantdji ) answered saying it is true. 

No idea if there is a workaround. Please post if you get it! 

Good luck!

rohitsantdji, 2018-07-06:
As @rcabg correctly says, there is no way to get video feed on M600 at the moment.

closed 2018-07-06 by rohitsantdji