dji-sdk/Onboard-SDK #343: From samples to real flight with onboard sdk

giueppo, 2018-06-25:
Hi everybody, I'm pretty new and I have spend a lot of time in trying to build onboard sdk with Eclipse rather than Keil.
Now, I would like to make an M600 flying.
How can I fly instead of running a sample in simulation?
Do I have simply to call flying function without opening simulation and not connect usb?
Can somebody provide me basic steps of main to perform a takeoff-hovering-landing procedure, after all the required initialization and so on?

Thanks!

zliDJI, 2018-06-27:
Yes, if you don't plug in the usb and run the simulator, your code will actually fly the drone. Please be aware that M600 is a big platform and flying without full confidence of your code in simulation environment can be very dangerous. Make sure you test in an open outdoor area, and start from the very simple actions like takeoff and landing.

giueppo, 2018-06-29:
Thank you. I imagined that...wow...luckly the usb cable was always in its position to save everything :)
I will let you know guys the next steps.
Of course we are going to test very simple things at the beginning in a outdoor area.
I imagine (and hope) that we can take always the control with the remote, don't we. We will check everything to be safe.
I keep this topic open for update as soon as I get the real flight.

giueppo, 2018-07-10:
I can confirm that it is sufficient to remove USB connections and everything works in real words.
I have some problems now concerning the simulation since I want to simulate my code, but motors failed to start. I opened another issue for this so I can close this one! Thank you everybody!

closed 2018-07-10 by giueppo