dji-sdk/Onboard-SDK #344: A3 takeoff failed in simulation mode

Jai-GAY, 2018-06-26:
Read App ID
User Configuration read successfully. 
STATUS/1 @ init, L55: Attempting to open device /dev/serial0 with baudrate 230400...
STATUS/1 @ init, L65: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L687: Device Serial No. = xxxxxxxx19
STATUS/1 @ parseDroneVersionInfo, L689: Hardware = A3
STATUS/1 @ parseDroneVersionInfo, L690: Firmware = 3.2.36.13
**ERRORLOG/1 @ initVirtualRC, L1029: Virtual RC is not supported on this platform!**
STATUS/1 @ activate, L1239: version 0x302240D
STATUS/1 @ activate, L1252: Activation successful
STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ initGimbal, L850: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=50, nTopics=1.
STATUS/1 @ removePackage, L461: Remove package 0 successful.
STATUS/1 @ initGimbal, L894: Gimbal not mounted!
| Available commands:                                            |
| [a] Monitored Takeoff + Landing                                |
| [b] Monitored Takeoff + Position Control + Landing             |
b
STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=10, nTopics=2.
**Motors spinning...
Takeoff failed. Aircraft is still on the ground, but the motors are spinning.**
STATUS/1 @ removePackage, L461: Remove package 0 successful.
STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=50, nTopics=2.
Task timeout!
STATUS/1 @ removePackage, L461: Remove package 0 successful.
STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=50, nTopics=2.
Task timeout!
STATUS/1 @ removePackage, L461: Remove package 0 successful.
STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=50, nTopics=2.
Task timeout!
STATUS/1 @ removePackage, L461: Remove package 0 successful.
STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=10, nTopics=2.
STATUS/1 @ getErrorCodeMessage, L644: 
STATUS/1 @ getCMDIDTaskMSG, L824: CONTROLLER_MOTOR_OFF
STATUS/1 @ reset, L560: Reset Subscription Successful.

what have i not done correctly ?

giueppo, 2018-06-29:
I am pretty new but I could make the simulation working with M600 and STM32. Are you using the same?
The only thing I notice is that: I have used lower baudrate, 115200, (still didn't used with the faster one).

Jai-GAY, 2018-07-02:
No, i am using [pi3 + A3 @230400bps](https://developer.dji.com/onboard-sdk/documentation/development-workflow/hardware-setup.html). the log mentioned Serial started successfully.

rohitsantdji, 2018-07-06:
Hi @Jai-GAY some additional questions:
- Does the simulation visualizer match the error messages thrown by the sample? i.e. is the aircraft sitting on the ground with the motors spinning?
- Can you connect DJI Go to the LB2 RC and make sure no errors/warnings/user input are visible when you run this sample code?

On another note, sending issues like this to dev@dji.com may get you faster response. We have recently set up a developer support program that will try and get questions like this routed to the right engineers for prompt solutions.

Jai-GAY, 2018-08-14:
@rohitsantdji , thanks, indeed, the SDK is very sensitive, must ensure IMU, Compass warning/error are not present to get the sample working.

closed 2018-08-14 by Jai-GAY