dji-sdk/Onboard-SDK #345: Firmware version not supported with M210ConfigTool

PaulKemppi, 2018-06-27:
Trying to issue --usb-connected-flight on command but I get:

Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/serial/by-id/usb-FTDI_TTL232R_FTA5ICGS-if00-port0 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L687: Device Serial No. = 095XE7D00200SR

STATUS/1 @ parseDroneVersionInfo, L689: Hardware = PM410

STATUS/1 @ parseDroneVersionInfo, L690: Firmware = 3.3.10.4

ERRORLOG/1 @ initVirtualRC, L1029: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1239: version 0x3030A04

STATUS/1 @ activate, L1252: Activation successful

STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ initGimbal, L850: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: 
STATUS/1 @ removePackage, L461: Remove package 0 successful.
STATUS/1 @ getErrorCodeMessage, L644: 
STATUS/1 @ getCMDIDSetControlMSG, L789: 
ERRORLOG/1 @ ConfigTool, L39: Please make sure the connected drone is a M210 and firmware version is supported.
Segmentation fault (core dumped)
-----------------------------

I have updated all the components of our M210 RTK and taken the latest version of this Onboard-SDK.

zliDJI, 2018-06-27:
Hi, @PaulKemppi   We will release a hotfix in a couple of hours to fix this version incompatibility issue. Sorry about that.

mabrowning, 2018-07-06:
Please upload arm7l binary!

rohitsantdji, 2018-07-06:
@mabrowning the armv7 binaries have been uploaded in b68c9ae. Apologies, slipped through the cracks the first time around.

closed 2018-07-06 by rohitsantdji

mabrowning, 2018-07-06:
Thanks! Works now.