dji-sdk/Onboard-SDK #348: Reset Factory issue DJI MANIFOLD ONBoard

BellFactory, 2018-07-07:
Hello. I try to reset factory of DJI Manifold by following this link below.
 https://www.skypixel.com/videos/manifold-tutorials-how-to-create-and-restore-a-system-image-mp4 
But I have an issue when I try to run this command "./nvflash --read APP system.img --bl ardbeg/fastboot.bin --go" I get this error as attachment picture. I have already check usb connection by command "lsusb" ,there is stiil "NVidia Corp.". 

Anyone Please help me. Thank you in advance
[image]

BellFactory, 2018-07-07:
Actually why I wan to reset factory manifold because I had wrong connection between A3pro dji m600 with manifold and it suddenly shutdown make me cannot boot manifold again.

JXZhou88, 2018-07-30:
Hi BellFactory,

i have encountered quite a similar problem with my Manifold in combination with Matrice 100. In my case, the Manifold also suddenly shutdown, after i was using it only for 1 hour after purchase! I also wasn't able to boot it after this incident.

I did successfully restore factory setting for my Manifold using the commands provided in your linked video tutorial, but this issue still persisted. Since you have encountered the same malfunctioning behaviour of your Manifold, i wanted to ask whether you have found a way to solve this problem. 

Regarding your Question, i think what the command "./nvflash --read APP system.img --bl ardbeg/fastboot.bin --go" does is actually making a back-up of the existing system image on the Manifold. Have you tried the command "sudo ./flash.sh jetson-tk1 mmcblk0p1" for restoring the settings?

closed 2019-05-31 by hummelslei