dji-sdk/Onboard-SDK #349: Failed to get drone version (M210 Onboard sdk)

azrimahanif, 2018-07-10:
Hi

We're having issues with the drone version when running the samples, the serial communication pass with no issues, but then error come in. 

[image]

there are also an issue with the dji assistant 2, where we couldn't get the drone latest firmware information. 

[image]

we have followed the instructions [DJI Onboard Software setup](https://developer.dji.com/onboard-sdk/documentation/development-workflow/environment-setup.html) here.

we have update the drone latest firmware using mobile phone. 
Is there any possible error during the hardware or software setup? 

Thank you in advance.

giueppo, 2018-07-10:
I had the same problem at the beginning. After checking thousand, I had some pin in wrong way.
You can check:
- if you inserted Activation code of your firmware
- if you activated the firmware by running it while the DJIGo App is opened and the device connected to internet
- during my trials, I also connect DJI go with the same account I registered the firmware.
- check again if the serial port are sending messages that make sense (look in exadecimal and look for packets)

rohitsantdji, 2018-07-10:
An additional check - after you updated your firmware using a mobile phone, can you go into Assistant's SDK page and make sure the Enable API Control box is still checked? Additionally, make sure that at least one broadcast data is set to a non-zero value on the same page.

azrimahanif, 2018-07-11:
Thank you @giueppo , @rohitsantdji ,

the activation is working, we re-install the dji assistant 2 and connect the controller with mobille phone during the activation.

Thank you so much for your help.

closed 2018-07-16 by rohitsantdji